martin-ger/esp_wifi_repeater #161: Source Code of Init.bin

vberkaltun, 2018-03-12:
Hi,

I want to make an init file that will connect directly to my own wifi modem. So, I guess I need to create a init file and combile it with open-sdk but I dont know that what does include init.bin file? Can you share the source code of init file which located in that repo?

...And additional, is there an any way to upgrade limitations of maximum connected stations of ESP8266?

vberkaltun, 2018-03-12:
I found something interesting in `esp_wifi_repeater/user/user_config.h` file. Can it work if we change the defalut value of `WIFI_SSID, WIFI_PASSWORD ,WIFI_AP_SSID` and others...?

```c
#define WIFI_SSID            "ssid"
#define WIFI_PASSWORD        "password"

#define WIFI_AP_SSID         "MyAP"
#define WIFI_AP_PASSWORD     "none"

//
// Define this to support WPA2 PEAP authentication (experimental)
//
//#define WPA2_PEAP	     1

#define MAX_CLIENTS	     8
#define MAX_DHCP	     8
```

martin-ger, 2018-03-12:
Yes, this will work.

marek26340, 2018-03-13:
But if you want to have more than 8 connected stations simultaneously, you are out of luck.

vberkaltun, 2018-03-13:
```c
#define WIFI_SSID            "esp8266core"
#define WIFI_PASSWORD        "11223344"

#define WIFI_AP_SSID         "esp8266mesh"
#define WIFI_AP_PASSWORD     "12341234"
```

I have compiled the `user_config.h` file as above, but it worked very very unstable. First of all, Even though I changed the value of `WIFI_AP_PASSWORD` and` WIFI_AP_SSID`, it worked as a public network.

Sometimes connected, sometimes did not connected to "esp8266core"...

\-

@marek26340 You mean in real life, right? Because It can be compiled above upper bound.

martin-ger, 2018-03-13:
Modify the defaults in config_flash.c. Set ap_open to false.