martin-ger/esp_wifi_repeater #162: Feature request - setting to limit data volume i.e. per day

heinerolli, 2018-03-14:
Is it possible to be able to set a limit for an i.e. dayly data usage.
(Not bandwidth, but actual data amount).
That would help me setting this s/w up for an Internet connection (through a router) via a 4G service.

martin-ger, 2018-03-14:
What should happen when the quota is excceeded? Block till next period?

Perhaps this cloud be done by a more general traffic shaping approach. Will think about it...

heinerolli, 2018-03-14:
In my use case "Block till next period" would be perfect!
Thanks for looking into this! Or thinking about it. :-)

heinerolli, 2018-03-30:
Hello martin-ger; 
do you think that you will consider adding this feature mid-term, i.e. in the coming months? 
I am asking, since this will rule, if I need to change my Internetaccess / -provider in our guest home or not.
All the best and happy easter

martin-ger, 2018-03-31:
Looked into it. Found two important questions:
- the start end end times, meaning: limit within 24 hours after startup, e.g. from 10:41 to 10:40 next day or starting at 00:00 till 23:59? The latter would require a real time clock (NTP).
- which traffic: upstream? downstrem? both? in sum?

heinerolli, 2018-03-31:
Thanks for your feedback! 
In a perfect world, it would be great, if it uses a ntp service - so that the user can not reset the traffic limit by just pulling the power plug .... 
But a simpler 24hr setup would already help a lot!
WRT to the 2nd question: a sum traffic limit would be good.

martin-ger, 2018-03-31:
Pulling the power plug will still work with NTP time as it resets not only time but also the traffic counter. 

BTW: do you have access to an MQTT broker?

heinerolli, 2018-03-31:
Ah - of course. Well, then the 24hr setup would be as good for me.

In the location, where this setup would be, there is no MQTT broker. 
The only hardware there is, is a "Speedport LTE" (or Huawei B593s-12), which connects via SIM to LTE.

Very simplistic setup....

martin-ger, 2018-03-31:
Added:
- set daily_limit _limit_in_KB_: defined a max. amount of kilobytes that can be transfered by STAs per day (0 = no limit, default)
- set timezone _hours_offset_: defines the local timezone (required to know, when a day is over at 00:00)
This should do the basic job. Reset also reset the Bytes_per_day counter.

heinerolli, 2018-04-01:
Martin - thank you!!! That is awesome. I will try it immediately. What a nice eater surprise. :-)

heinerolli, 2018-04-01:
Martin - I tested it and it works great!! Thanks again for including this feature in your already great software.
I am currently testing it, therefor these questions:
When I set the time zone, I do not see that change reflected in the 'show stats' quoted time. Is that intended?

martin-ger, 2018-04-01:
It should. Set it, save, and reset.

heinerolli, 2018-04-01:
Yes, you are right - I earlier tested it directly after setting it. After save & reboot, its updated. Thanks again!!

heinerolli, 2018-04-04:
So I continued to test it and I see a weired behavior:
My "newer" devices i.e. Windows 10, Intel Core I3 work w/o problems connected through the ESP repeater. I get high throughput of around 5kbps in both directions. But my older devices (IPhone 4s and an Intel Atom based Win10 Tablet) show major problems - they hardly can open a webpage & I get around 100kbps(!) throughput when tested via i.e. wieistmeineip.de. The same devices work w/o problem when connected directly (via router) to the internet.  Can you make sense out of this? 
Otherwise - great software and great use case for the ESP's!

heinerolli, 2018-04-05:
Unfortunately, the "very low bandwidth issue on slower devices" is still present...it got better, but its not gone. I.e., on the old Iphone, I can not access "kundenkonto.lidl-connect.de" via the ESP-AP. When using the direct WLan access to the router, I have no problem accessing the same site w/ the phone. (Also, as earlier stated, using faster PCs, I can access the site also via the ESP).
Some observations, which might help pinpoint the problem:
- The issue seems to be getting worse over time; being best directly after rebooting Router & AP
- I noticed that my ping times are in general pretty high (300ms) over the 4G network my router uses for the internet. Can this create problems for the ESP-AP?
- After longer times of idle (i.e. 2 hours) my connected devices loose internet connection - they are still connected to the ESP. When Windows "fixes" the problem, it states that it fixed an IP conflict.
Can you give me some guidance on how to troubleshoot this issue?
Finally a question for understanding: When I try to set the WLan mode to 1 (=b), "show stats" still states phy mode n.  (After setting phy mode, the device reboots, so I can not do a "save &reset") "Show stats" also still states phy mode n after I set my router WLAn to b mode. So: What am I doing wrong? In general: Can the ESP be working as an AP,in one mode while the Router it is connected to uses a different mode?? 
Sorry for pestering you with all these questions - I really would like to get it to work!

martin-ger, 2018-04-05:
Let me try to answer at least some of the questions:
- The idle issue is probably a DHCP timeout: look into "show dhcp" this will tell you the remaining time of a lease in minutes (currently a initial lease is 2 hours = 120). When a lease runs out, another device might claim the same IP.
- phy_mode setting is actually somewhat tricky: it only works when you disconnect all WiFi interfaces ("set ap_on 0" and "disconnect") now you can set the phy_mode and save it. However, I havn't found any use case for changing the phy_mode (added it on request): In StationAP mode, the mode is g anyway (n doesn't work with the chip) - forcing it to b is probably no good idea...
- One problem I know from (older?) Android Smartphones is, that they think it would be a good idea to flood the AP for several ms with link-layer packets before starting web-traffic. This is a high load for the few buffers of the ESP and might lead to packet loss especially at the start of a web session.
- Also, after some time of heavy traffic the NAT tables become full - this might terminate some TCP connections

heinerolli, 2018-04-06:
Thanks for your explanations, Martin. Is there a way to check if the NAT table is getting full / or if the buffer is getting full / or if packets are lost? Some kind of debug mode? 
Or other troubleshoot ideas?

heinerolli, 2018-04-06:
Also, I need to revise my earlier assessment: Its not only slow devices..... Now that I switched these off / connected them directly to the router, and work on the faster machine, that also slows down .... Simple Webpages like Wikipedia home take forever to load. But it fluctuates: It seems to be able to recover on its own. (its not crashing / rebooting: 'show stats' shows the corrects uptime since last intentional boot.

martin-ger, 2018-04-08:
Do you have the serial connected? Some output "Tx..." there? This is a sign of no buffers left.
Currently the is no debugging of the NAT tables available on the UI (will think about that)
The help issue is probalbly because the output became longer than the output buffer. (Will check this as well)

heinerolli, 2018-04-08:
Hi Michael; thanks for having the patience and following up...
I connected now serial (should have done that earlier - sorry...) and i get maybe something interesting:
I get the message "LmacRxBlk:1"
When I only have 1 ("faster") device connected, this comes very rarely.
If I connect (only 1 ) "slower" device, I get this immediately a lot (i.e. ~10-20 times when opening a webpage. 
(So, for testing in both cases only 1 device is connected to the ESP)
Does this tell you something? Hopefully....
Have a nice sunday and thanks again!

martin-ger, 2018-04-08:
"LmacRxBlk" - That's what I meant - Buffer issue... And I think, this is esecially caused by these devices that send out this flood of packets (actually Firefox - is it Firefox mobile?):
 
https://marius.bloggt-in-braunschweig.de/2017/07/05/android-firefox-und-das-wifi-tickling/;
"If you’re on wifi and an IPv4 DHCP network we will send 0 length UDP packets at port 4886 of your gateway at the default rate of 60hz for 400ms from the start of the transaction in an attempt to improve RTT during the critical early phases. I call this „tickle time“

Try to set an ACL "acl from_sta UDP any any any 4886 deny". This cannot avoid these packets (you only can avoid them by reconfiguring Firefox).

martin-ger, 2018-04-08:
Limiting the bandwidth cannot help: it simply drops (more) packets when the budget is empty...

heinerolli, 2018-04-09:
Thanks Martin.
I am not using firefox, but tried the suggested acl rule anyway. Unfortunately, applying that rule blocked the complete Internet access! But maybe something with that rule is wrong? show acl states 0 hit, but still many packets blocked (I also tried to_sta):
"From STA:
UDP any:any any:4886 deny (0 hits)
Packets denied: 1763 Packets allowed: 138"

Aber hoffentlich sind wir auf der richtigen Spur. Dazu eine weitere Frage:
Womit / wie kann ich solche wie im verlinkten Artikel gezeigten packets im WLAN tracken:
16:05:52.958119 IP 192.168.20.3.59077 > 1.0.0.0.4886: UDP, length 0
16:05:52.973897 IP 192.168.20...
Damit könnte ich sehen, was das Problem auslöst. Denn es scheint kein generelles Bandbreitenproblem zu sein - ich kann z.B. (von dem "schnellen" Rechner) problemlos Skypen über den ESP - ohne das "LmacRxBlk" ausgelöst wird.

martin-ger, 2018-04-09:
Sorry, you should add another rule afterwards:
"acl from_sta IP any any allow"

Man kann die Pakete mit dem monitoring beobachten: also "monitor on 5000" und dann auf einem (Linux-)Rechner auf Uplink-Seite: "netcat [external_ip_of_the_repeater] 5000 | sudo wireshark -k -S -i -" 
Dann kann man zumindest jedes Paket sehen, dass zwischen ESP und Clients läuft und empfangen wird.

heinerolli, 2018-04-09:
Oh - ich hab' aus versehen auf deutsch geschrieben. Scheint aber auch zu gehen. :-)
Also, ich hab die zusätzliche Regel eingesetzt. Damit geht der Internetzugang wieder - aber die Fehler kommen auch wieder.
Ich würde das mit dem beobachten der Pakete gerne versuchen - habe hier allerdings vor Ort nur  Windowsrechner zur Verfügung.....

heinerolli, 2018-04-09:
Ich hab jetzt was interessantes gemerkt:
Ich habe bisher immer von unterschiedlichem Verhalten von 'langsamen' / 'schnellen' devices geredet.
Das Problem ist aber ein anderes: Der Tablet PC ('langsam') am ESP generiert Probleme (-> 'LmacRXBlk:1'), wenn er im Akkubetrieb läuft - er tut das nicht, wenn er am Powersupply hängt! Ähnliches passiert am Laptop ('schnell').  (Bei beiden devices finde ich keine Energiespareigenschaften, die das WLAN betreffen, womit ich das Verhalten beeinflussen könnte...)
Der Wireshark output ist für mich leider erstmal viel gibberish - Es scheint aber, das mehr 
"TCP retransmission" und 
"TCP Previous segment not captured" und 
"TCP Out-of-order" 
vom ESP zum device (und umgekehrt) auftritt im Batteriebetrieb.
Haben sie hierzu irgendwelche Ideen?

P.S. Das IPhone generiert immer den o.g. Fehler, egal ob es geladen wird oder nicht.....

martin-ger, 2018-04-10:
Interessante Beobachtung, aber ich kenne keine unmittelbare Begründung, was die Energieversorgung mit der WiFi-Übertratung zu tun hat, außer ggf. Sparmodi. Das könnte wiederum zur Bursts führen, die ihrerseits dann vielleicht eher zum Overflow führen als eine kontinuierliche Übertragung - reine Spekulation. Kann vielleicht aus dem Timing der Pakete erkennen.

Dass mehr Retransmissions und Out-of-Order auftreten ist sicher eine unmittelbare Folge von Paketverlusten.

heinerolli, 2018-04-10:
Ich habe in Wireshark die Paketstatistiken eines gleichartigen Versuchsablaufes (cache löschen, tagesschau aufrufen) angeschaut und ich kann keinen Unterschied der Paketstatistiken feststellen: Ähnliche Verteilung der Paketgrössen, ähnliche Burstrate, ähnliche Anzahl der Pakete pro ms. Ich begreife es nicht.
Gibt es irgend eine Möglichkeit die Buffer-Grösse des ESP-AP zu verändern?

heinerolli, 2018-04-10:
Hallo Martin; ich glaube, ich habe den root cause gefunden: 
Es hat definitiv mit dem Energiemanagement der Verbundenen PCs zu tun: 
Nachdem ich bei meinem "langsamen" Tablet alle Drahtlosadapter-Energiesparoptionen ausgeschaltet habe, funktioniert es wunderbar über den ESP! (Wenn die Auswahl-Optionen wie bei mir nicht vorhanden sind, kann man die über regedit Eintrag hinzufügen - siehe https://www.tenforums.com/tutorials/73119-add-remove-wireless-adapter-settings-power-options-windows-10-a.html)
Ich habe zu dem Problem auch eine Erklärung von MS gefunden:
"You may experience random connectivity issues when you connect a Windows-based mobile PC to certain Wi-Fi "hot spots." These connectivity issues include the following:
- The wireless network connection is dropped.
- You experience poor performance.
You experience these issues if the computer is running on battery power. 
Cause:
This issue occurs if the Wi-Fi hot spot uses wireless APs or routers that do not support the 802.11 power save protocol." 
(https://support.microsoft.com/en-us/help/928152/you-may-experience-connectivity-issues-or-performance-issues-when-you)
Das heißt, das Problem könnte evtl. generell gelöst werden, wenn man dem ESP-AP das "802.11 power save protocol" beibringen könnte. (Denn z.B. bei Mobile Phones ist das Problem sonst nicht behebbar, da dort WLan-Powersave-Optionen nicht geändert werden können.)
Wäre das möglich für Dich? Es wäre m.E. eine deutliche Funktionsverbesserung für Dein Programm / den ESP-AP. Nicht nur für mich....
Ich bin gespannt auf Dein Feedback

martin-ger, 2018-04-10:
Okay, vielen Dank für die Analyse und die Beharrlichkeit dabei, das ist in der Tat interessant auch für andere Nutzer.

Die bad News ist, dass ich daran, selbst, wenn ich es wollte, praktisch nichts ändern kann:
- die IEEE802.11 Implementierung ist Closed Source (auch womöglich aus Gründen um die FCC-Zertifizierung bekommen zu können)
- es bedarf aber Änderungen hier: man müsste u.a. Control-Frames senden können, was mit dem SDK nicht geht. Auch das Timing ist kritisch hier und dazu kommt noch die Tatsache, dass der AP Frames zwischenspeichern muss - und das, wenn der Pufferplatz eh schon das Problem ist.

Also sehe ich hier zz. keine Möglichkeit, wenn man nicht den Endgeräten beibringen kann, dass zu lassen...

heinerolli, 2018-04-10:
Das ist wirklich schade! Kann man bei Espressif irgendwie jemanden darauf hinweisen? Das wäre sicher auch für die interessant - und vielleicht auch ein Verkaufsargument für den ESP32, wenn sie es dafür fixen würden. Jedenfalls vielen Dank, dass Du nicht genervt warst von meiner Hartnäckigkeit.... :-) Und danke für die S/W, die erstaunliches aus dem 8266 herausholt!

heinerolli, 2018-04-10:
Hallo nochmal. Entschuldige meine Ignoranz - ich möchte das mit der closed source Geschichte verstehen: Bei der SDK, die Du in der Anleitung verlinked hast, steht u.a "OpenSource components of the SDK are based on: ...net80211/ieee80211 (FreeBSD WiFi stack), http://www.unix.com/man-page/freebsd/9/NET80211 (source withheld by Espressif)" Und auf der freebsd net80211 Seite gibts ein setting:
"IEEE80211_IOC_POWERSAVE
	     ... powersaving mode in i_val.  Valid values are
	     IEEE80211_POWERSAVE_OFF (power save operation is disabled) and
	     IEEE80211_POWERSAVE_ON (power save operation is enabled)." 
Wäre das setting nicht irgendwie in der SDK ansprechbar?
Wie gesagt - entschuldige, wenn das alles keinen Sinn macht, was ich schreibe. Ist alles neu für mich....