martin-ger/esp_wifi_repeater #163: Compile Warning on "gpio_intr_handler"

geocheats2, 2018-03-15:
```
user/user_main.c: In function 'gpio_intr_handler':
user/user_main.c:2884:6: warning: passing argument 3 of 'ets_timer_setfn' makes pointer from integer without a cast [enabled by default]
      os_timer_setfn(&inttimer, int_timer_func, easygpio_inputGet(USER_GPIO_IN));
      ^
In file included from user/user_main.c:4:0:
/home/user/esp-open-sdk/sdk/include/osapi.h:67:6: note: expected 'void *' but argument is of type 'uint8_t'
 void ets_timer_setfn(os_timer_t *ptimer, os_timer_func_t *pfunction, void *parg);
```

my gpio_in_int works only once could this be the problem, sorry for the idiotic issue but i don't really know enough programming to repair that

martin-ger, 2018-03-15:
The third parameter is a generic pointer to any kind of argument (structure). I use it just to pass 0 or 1. Not nice, but should not be a problem.
What do you mean: only works once?

geocheats2, 2018-03-15:
really sorry for wasting your time i am a total idiot, i had int re-enable commented out.

closed 2018-03-15 by geocheats2