martin-ger/esp_wifi_repeater #165: common network  on both sides

joe-ger, 2018-03-17:
Hello Martin,
 first thanks for the job you did so far, well done!

For a  special use case I need to have the  some ip addresses  for  clients, if I'm directly connected to my  router or go other your NAT repeater.  The changes  I did  so far are currently mainly in the LWIP library.
 The  only  modification I did  in your repeater  code itself  at  a ping call to the  sta_watchdog.
Are  you  interested  in these  modification?

Regards
Joe

martin-ger, 2018-03-18:
Joe, I don't think I understand exactly the setup, resp. the modifications. Could you give me an example?

joe-ger, 2018-03-18:
Hallo Martin,
 wenn Sie nichts dagegen hast schreiben ich auf  Deutsch.

 Ziel  der  Veränderungen war es,  den NAT ROUTER  in nur einem Subnetz einzusetzen.
 das heißt auf  beiden Seite den selben Addressbereich zu haben
 Ich  habe folgende Veränderungen gemacht
 DHCPserver um ein forward  Funktion erweitert so  das  der DHCP Server sich wie ein DHCP relay agent  verhält.

ip.c
 in ip_input   forward  Pakete  auch  wenn sie  zum gleichen  Subnetz gehören udn dann im ip_forward  kreuze die Interface.

in udp
 kleine  Änderung  um zu  verhindern das das   dhcp forward  auch  im Router subnetz  aktive  ist
 
 und schließlich  in etharp.c
 Erweiterung  der Funktion etharp_arp_input  so  das  sie  die richtige  MAC Adressen zurückgibt.
 Hier noch ein paar  kleinere  Änderungen bzw   Entfernen von Preprocessor  swiches  nötig.

  Das  sind im groben die  Änderungen. Sie sollen im Normalbetrieb keinen Einfluss haben.

  was user.c betrifft   habe ich  nur  das  folgende gemacht. 
 ich  schicke den bereits impementiere  ping  Befehl  wenn der  sta_watchdog_cnt  auf  15   steht.
 und  setze  ihn auf 20  wenn  sich  ein client  vom repeater  abmeldet.

 Ich hoffe das hilft.  Ich kann aber auch zu jeder Zeit ein Diff machen und Ihnen schicken.
 Die Änderungen habe ich gegen Ihre Version 1.7.4 gemacht.

 Joe

martin-ger, 2018-03-18:
Klar, Deutsch ist prima! ;-)

Ich würde mir das auf jeden Fall gerne ansehen und würde mal schauen, was man da mainstream nehmen kann. Könnten Sie nicht eine Fork machen und kann man die Diffs ja sehen und auch gleich testen?

Martin

joe-ger, 2018-03-18:
Hallo  Martin, 
ich arbeite zwar seit kurzen beruflich  mit GIT möchte aber nix kaputt machen. 
Konfigurationkontrolle und dann alles verschossen ein lediges Thema.
Mit github hab ich nun noch gar keine Erfahrung, daher wären ein paar Stichworte hilflich.

 Wenn ich  dich  richtig  verstanden  habe soll in eine Branch  generiren und dort  meinen Kode  einchecken- korrekt?

 Joe

martin-ger, 2018-03-18:
Genau. Bei mir kaputt machen geht eh nicht. ;-)

Mein Repository als eigenes forken, das dann mit "git clone ..." lokal kopieren. Dort ändern und dann mit git add ..., git commit ... und git push origin master wieder hoch laden. Dann können alle (und wir sehen) und ggf. machen wir einen "Pull request" und ich kann etwas übernehmen.

martin-ger, 2018-03-18:
Ich habe gerade die Änderungen im lwip gesehen.Das sieht interessant aus - was genau machen die Änderungen in etherarp.c? Kriegen wir damit doch noch einen echten Repeater hin?

joe-ger, 2018-03-18:
Ich habe einen Fork erzeugt und die Äanderungen eingecheckt, sind insgesamt 5 Dateien. Da die  Änderung  im user-main.c minor sind und nur den Ping betreffen hab ich sie noch nicht eingecheckt 
( Der Grund das  ich es übehaupt gemacht habe, war, wenn kein klient  verbunden hatte, löste watchdog  sändig aus, da  kein Traffic  erkannt  wurde. ist  verleicht  auch ein Denkfehler  meinerseits).

In meinem Set-up hat der Router im Moment  die xxx.yyy.zzz.2  daher konnte  ich  im Repeater mit der xxx.yyy.zzz..1 arbeiten. Diese IP wird auf beiden Interfaces des Repeaters verwendet. Eine Möglichkeit  wäre hier, das wenn beiden Netzwerke identisch  sind,  entweder  die STA zu  übernehmen oder noch  einen Parameter für eine IP Address einzufügen. Ich  würde für ersters plädieren damit es nicht zu IP-Kollisionen führt, ggfs  nur  mit Flag STA  übernehmen.
 
 2 Anmerkungen noch :
1. Ich habe das NAPT auf false gesetzt so das die Pakete nicht manipuliert werden, spart mir einen Extra check.
 ich  mußte  das  LWIP_MDNS  flag  uaf  0  stzen ansonsten  funktionierte  das  DHCP  nicht  mit  einem Windows  Laptop. Bedingt durch die Problme, die ich in diesem Bereich hatte, habe in im Moment eine feste IP im Downlink eingestellt. Werde das in den nächsten wieder auf DHCP umstellen und testen.

 Joe

martin-ger, 2018-03-18:
Der Watchdog sollte gar nicht auslösen, solange er auf "none" (= default) steht - tut er trotzdem?

joe-ger, 2018-03-18:
Ich denke, wir sollten noch  einen echte Repeater hinbekommen, werde es die  Tage noch einmal prüfen. Wir sollten in diesen Fall nur reinlaufen wenn beide Netzwerke gleich sind ansonsten nicht us_proxy ist flase solange nicht  netzwerke gleich sind. Wenn ich es  richtig  gesehen habe wird  die  netif  information nur  bei nmtp  udn nun bei  mir  verwendet.

 Zum Watchdog ich hatte  Ihn auf  5 min 300 sec gestellt.  Die seriellen Verbindung angeschlossen und und mit  dem  Router verbunden, danach löste der watchdog alle  5 Minuten aus.  Der Grund für die  Verwendung des Watchdog, war dass ich Problem mit  der Verbindung hatte,  wenn ich mit meinem Windows  Rechner  zwischen den  Netze hin und her geschaltet habe.

joe

joe-ger, 2018-03-18:
Noch eine Anmerkung  zum Watchdog. Er  wird ordnungsgemäß zurückgestzt  wenn ein  Verbindung  zu  einem Client  besteht.

Korrigiere  mich  wenn ich  deinen Kode nicht  richtig  verstanden  haben sollte: 
Du ersetzt   beim Connect  durch  den ptach_netif  Befehl  den orginal  aufruf für  die sta_input mit  my_input_sta  danach  ruft  deine Funktion  die  orginal  auf  das  machst  du  dann nochmal wenn sich  ein Client verbindet, kann es sein das dort etwas ins Leere  läuft, wenn sich ein Client disconnected oder kein Verbindung existiert? 
joe

martin-ger, 2018-03-18:
Die Idee ist: die input-Funktionen beider Interfaces werden dauerhaft ersetzt/umgeleitet, sodass ich alle Pakete sehe. Das ist für das Forwarding und das NAT völlig egal, damit kann ich aber Mitschneiden (Monitor), die ACLs realisieren, Statistik führen und eben auch die Watchdogs. Wenn die keine Pakete mehr sehen. egal, ob es keine Verbindung gibt oder die Verbindungen/STAs nichts sendet, wird der Reset ausgelöst.

joe-ger, 2018-03-18:
Das klingt sinnvoll, meine Annahme war/ist, es sollte doch in 5 Minuten irgendeine Kommunikation zwischen Router und Repeater stattfinden und damit den Counter zurücksetzt werden. 

Wenn ich mit Recht errinnere, habe  auch  versucht  den Repeater zu  pingen, dies hat den Watchdog  auch nicht zurückgesetzt.  
  
Werde mir das nochmal im Detail anschauen.

joe-ger, 2018-03-19:
Hallo Martin,
ich  hab mir die Datei  etharp.c nochmal angeschaut 
Du hast Recht die Zeile :  
for_proxy = (u8_t)ip_addr_netcmp(&dipaddr, &(netif1->ip_addr), &(netif1->netmask)); 
ist  unsauber und könnte  zu  Problemen führen 
 Richtig ist auf  beide Interface  zu  prüfen. 
 
for_proxy = (u8_t)(ip_addr_netcmp(&dipaddr, &(netif1->ip_addr), &(netif1->netmask)) & ip_addr_netcmp(&dipaddr, &(netif->ip_addr), &(netif->netmask)) );
 damit geht er in den Branch  for_proxy  wirklich nur wenn beide Subnetze gleich sind.  

 Joe