martin-ger/esp_wifi_repeater #166: Reset to default

khattak97, 2018-03-17:
Sir, great work.
Here I need your kind help for managing a physical push button which can restore it to default settings.
I tried but didn't work.
I grounded reset terminal but wasn't working even tried from web interface .
Please help....

martin-ger, 2018-03-18:
To do a factory reset by software, use the serial port, a terminal program at 115200 bps and enter "reset factory".

If you want to do it with a HW button, the easiest way would be to add some logic in the "timer_func()" in user_main.c. Check a GPIO pin there and if it high/low let's say four times in a row (at least 2 secs), do

```
config_load_default(&config);
config_save(&config);
blob_zero(0, sizeof(struct portmap_table) * IP_PORTMAP_MAX);
os_printf("Restarting ... \r\n");
system_restart();
```
Does this help?

martin-ger, 2018-03-18:
Added that in the latest version. Pull GPIO 0 low for at least 2 secs and the ESP will restore factory default values and reboot.

You can change the GPIO no in user_config.h.

closed 2018-03-18 by martin-ger

khattak97, 2018-03-18:
Thank you so much Sir, let me give a try to the new version.

martin-ger, 2018-03-18:
Caution: I changed the GPIO to 12 to avoid any confusion with the pin 0 pulled low for flashing.

khattak97, 2018-03-18:
Actually I'm working on Esp01 there is only GPIO 0 and 1.
Isn't it possible on GPIO 0 or 1?

martin-ger, 2018-03-18:
It is possible - you can adapt it to any GPIO in user_config.h. If you don't want to compile it, try the binaries from the intermediate version I pushed earlier in the morning:

https://github.com/martin-ger/esp_wifi_repeater/tree/c083d58e6e80cac29403811d3245d64d80d239e3

Here I compiled it with GPIO 0 as reset.