martin-ger/esp_wifi_repeater #168: How I Create bin files for flashing from this Source code?

rmhtmlkoodaaja, 2018-03-20:
Hello,

I Translated this firmware's HTML code to Finnish  and I want build bin file for sharing. How I do it?
I have installed Linux on my PC but how can I build bin file 0x00000 and 0x100000? Like in source code's folder firmwares?

RM_HTMLKoodaaja

martin-ger, 2018-03-20:
Follow these instructions to build then software: https://github.com/martin-ger/esp_wifi_repeater#building-and-flashing

rmhtmlkoodaaja, 2018-03-20:
Hello, 

I did it, but it won't work for me. If I send you my edited source code via e-mail, can you compile it for me?

martin-ger, 2018-03-20:
Okay, but only once ;-)

rmhtmlkoodaaja, 2018-03-21:
Thank you! I will send you source code today. I'll fix problem of building bin files later.

rmhtmlkoodaaja, 2018-03-21:
Hello,

I uploaded my edited source code to Google Drive as ZIP file.

https://drive.google.com/open?id=1wbS-oaaTJBRRQp14OCN0X3YNik57DYje