martin-ger/esp_wifi_repeater #169: A bug in add static routing

SmartChild2018, 2018-03-22:
I'm very poor in English. I hope you can understand it.

The effectiveness of routing is not checked when add static routing
>
    CMD>show route
    Routing table:
    Network              Dest
    192.168.4.0/24       ew1
    192.168.10.0/24      ew0
    default              192.168.10.1
    CMD>route add 10.0.2.0/24 192.168.1.20
    Route added
    CMD>route add 10.0.2.0/24 192.168.1.21
    Route added
    CMD>route add 10.0.2.0/24 192.168.1.22
    Route added
    CMD>show route
    Routing table:
    Network              Dest
    10.0.2.0/24          192.168.1.20
    10.0.2.0/24          192.168.1.21
    10.0.2.0/24          192.168.1.22
    192.168.4.0/24       ew1
    192.168.10.0/24      ew0
    default              192.168.10.1

martin-ger, 2018-03-23:
Will check this with the next release and removed a conflicting entry before adding a new one.

martin-ger, 2018-03-26:
Fixed with latest version

closed 2018-03-26 by martin-ger