martin-ger/esp_wifi_repeater #175: can't connect to ssid with space character

parsamrz, 2018-03-31:
hi i have problem connecting to ssid's that have space in their ssid names for Ex "mrz ssid" please help me Thanks.

HowToMeetLadies, 2018-03-31:
Hello. He who seeks shall find: There are 6 issues about that topic already. See #154, #132, #108, #85, #75 or #48... 😉

Console: Spaces need to be escaped, so prepend a backslash. Web: Use %20 instead of space. 

@martin-ger: Vielleicht mal ganz oben und besser noch in fett+rot in die README nehmen? 😄

martin-ger, 2018-03-31:
Use %20 instead of space.

closed 2018-03-31 by martin-ger

xmagickx, 2019-08-31:
Sorry for commenting. I've read all the issues... I've a unique problem... My AP contains an "APOSTROPHE"... Please suggest a suggestion for this...

 
Shyam's WiFi

How to connect to this by esp8266 wifi repeater module? What characters to replace in place of APOSTROPHE? Even after using \ in place of space it shows + when the name of ssid is given

672Asimk, 2021-09-05:
i did use "Air\ Ap" in cli . it save sucessfully save the ssid but cannot connect with "Air Ap" ssid. in arduino serial moniter its only show "Air" not "Air Ap".

martin-ger, 2021-09-05:
Strange, should work...
Try %20 instead of space.

junaidpmr, 2022-02-23:
ssid : wifi
Password: 1234567
"192.168.4.1/?ssid=wifi&password=1234567"

Above ssid and password is connected
Problem is ssid: wifi ap
password: 1234567
"192.168.4.1/?ssid=wifi+ap&password=1234567"
Not connected
"192.168.4.1/?ssid=wifi%20&password=1234567"
Not connected
How to connect ssid with space character

fuzunspm, 2022-12-05:
did you find any solution?

martin-ger, 2022-12-05:
Please read the Readme.