martin-ger/esp_wifi_repeater #177: Feature update request: Auto connect delay

marek26340, 2018-04-05:
Hello! 
Is it possible to set a delay in auto_connect? Like "set auto_connect 5" so it will try to connect again in 5 seconds...

martin-ger, 2018-04-05:
"autoconnect" is a feature in the closed libs - I don't know how to change that timeout (other than disabling it and doing it manually).

Why do you think, this is neccessary?

marek26340, 2018-04-05:
I'm asking for this feature, because when the ESP is searching for it's uplink AP, you cannot modify settings using telnet without dropouts (WiFi very unstable, so is telnet.). If WiFi isn't lagging at the moment, then it happens that I start sending data through telnet, but the ESP starts scanning in the middle of transmission and ESP returns "Invalid command" after the scan is done.

martin-ger, 2018-04-05:
Ok, I understand - however, as said: think, it has to done manually with a timeout...

marek26340, 2018-04-05:
So this is impossible to do? Or it's possible?

martin-ger, 2018-04-05:
Guess, it is possible - but somebody has to do it... ;-)

marek26340, 2018-04-05:
well then, who is that "somebody"?

marek26340, 2018-04-11:
Aaaaaand you are gone again....

HowToMeetLadies, 2018-04-12:
Hello @marek26340,

Please do not get nervous. Do not get me wrong but feel free to fork this repository and implement that feature - otherwise you have to wait until it's done. 😉

@martin-ger Excuse me answering here, but my phone is ringing all the time - no need for meaningless messages.

Greetings
Chris

marek26340, 2018-04-14:
I'm not getting nervous :) 

@martin-ger are you at least going to tell me/us if this feature will be available?

hydracrux, 2019-03-09:
I resolve this problem putting a count in the 500 Ms interrupt and count to about 10 minutes. Then call connect function. And disable the auto connect function.