martin-ger/esp_wifi_repeater #178: how to define allowed clients?

giovanne123, 2018-04-09:
I want to configure / allow only specific clients to connect to the ESP-Wifi-Router.
I there a possibility to do this? (e.g. config MAC adresses of allowed clients)

Reason: I have a lot of static clients in my local net (subnet1 192.168.0/24). Eg. client with 192.168.0.49
And if such a client connects to the ESP (subnet2 192.168.1/24) he gets an IP in subnet2 range 192.168.1.x and isn't reachable under the 192.168.0.49 anymore.  

Is there a possibility to allow only defined clients?

martin-ger, 2018-04-09:
No, I don't know a way do disallow a STA to connect to the AP. In fact, I could deny to hand out an IP via DHCP, but it still will connect and wait for a long time for an IP.

If you want to prevent stations, why don't you configure 2 different SSIDs?

giovanne123, 2018-04-09:
Thanks for your answer.
I was hoping that there maybe is a possibility. There are so much options in ESP-Wifi-Router, and I dont know all ;)

Yes, a different SSID I think will be the solution I have to choose. Hopefully my clients are able to configure two Wifi APs with priority...

closed 2018-04-10 by martin-ger

giovanne123, 2018-04-10:
one question, how is this handled on a reapter fromm e.g. AVM connected to AVM fritzbox, where all the clients gets ip adresses in the main/one subnet?

martin-ger, 2018-04-10:
These repeaters are real "Repeaters", i.e. layer 2 devices. My Implementation is in fact a Router, i.e. a layer 3 device that does routing instead of pure switching. Routing requires IP (sub-)networks.

Haven't managed to realize a real repeater so far, guess it requires more access to the drivers.

giovanne123, 2018-04-10:
OK thanks.