martin-ger/esp_wifi_repeater #179: NodeMCU never boots up esp_wifi_repeater again after "set web_port 0", save and reset

marek26340, 2018-04-14:
Pretty much everything is explained in the title of the issue. 
I erased my ESP with esptool, reflashed the newest esp_wifi_repeater with the binary files + esp_init_data, configured STA, AP, CPU speed to 160, hostname, config port, config access. Saved and rebooted. All is OK, but when I set web_port 0, save reset, then the ESP never boots up again!
But I was thinking that I configured something wrong before, so I erased again, reflashed esp_wifi_repeater + esp_init_data_v8_vdd33 bin files. Booted up perfectly fine. Now that I have a clean installation, I immediately tried to set the web_port to 0, save, reset and........it stopped working again.

Here's the serial output after setting web_port 0, save + reset: 
`WiFi Repeater V1.8.1 starting

Config found and loaded
Starting Console TCP Server on port 7777
mode : sta(60:01:94:51:a7:c1) + softAP(62:01:94:51:a7:c1)

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x40100000, len 31824, room 16 
tail 0
chksum 0xea
load 0x3ffe8000, len 2064, room 8 
tail 8
chksum 0x62
load 0x3ffe8810, len 6040, room 0 
tail 8
chksum 0x45
csum 0x45
⸮⸮⸮⸮o⸮{⸮⸮o|⸮dddd`c⸮⸮|{⸮d⸮o⸮⸮n⸮`
Also after setting web_port to 0, I am unable to do "reset factory", it just doesn't respond to that command. HW reset also does not work.

Before setting web_port to 0, everything is normal.

martin-ger, 2018-04-15:
Can confirm this prob. Did a quick fix, but do not really understand, why this happens.

marek26340, 2018-04-16:
Thanks for fixing! I will try it again as soon as possible. If the fix worked, I will close this.

marek26340, 2018-04-17:
Looks like the issue is fixed. Have you figured out what went wrong?