martin-ger/esp_wifi_repeater #182: reboot loop: rf_cal[0] !=0x05,is 0xFF

torntrousers, 2018-04-17:
After flashing I get a rebooting loop with the below output. I've got the blinky example from the open sdk working, and esp_wifi_repeater appears to build ok.

I did get a similar problem with the blinky example but that was fixed by doing

```esptool.py -p /dev/cu.SLAB_USBtoUART write_flash 0x3FC000 ../../../esptool/test/images/esp8266_sdk/esp_init_data_default.bi```

But now 

```esptool.py --port /dev/cu.SLAB_USBtoUART write_flash -fs 4m -ff 80m -fm dio 0x00000 firmware/0x00000.bin 0x10000 firmware/0x10000.bin``` get the rebooting loop.

I've downloaded esp_init_data_default_v08_vdd33.bin and tried various ```esptool.py -p /dev/cu.SLAB_USBtoUART write_flash 0xFC000 ../esp_init_data_default_v08_vdd33.bin``` and also to 0x3FC000, but always just the  rf_cal[0] !=0x05,is 0xFF loop.

Any ideas?

```
load 0x40100000, len 31824, room 16 
tail 0
chksum 0x1d
load 0x3ffe8000, len 2064, room 8 
tail 8
chksum 0x24
load 0x3ffe8810, len 6044, room 0 
tail 12
chksum 0xf1
csum 0xf1
rf_cal[0] !=0x05,is 0xFF
```

marek26340, 2018-04-17:
I've never seen that, I usually see garbage like "csum err" or "Fatal exceptiop"........ Try erasing the flash memory with 

> esptool.py -p (your ESP COM# port) erase_flash

and try to upload again.........

martin-ger, 2018-04-20:
What kind of board is it? Mem size?

torntrousers, 2018-04-20:
This:

[image]

On the back it says ESP12E DEVKIT V2, www.doit.am

marek26340, 2018-04-20:
I have the exact board. It's an ESP8266 with 4MB of flash, CH340 serial to UART converter, AM1117 3.3V voltage regulator. 
I'm using the NodeMCU flasher for...well...flashing the ESP. 0x00000.bin to 0x000000, 0x10000 to 0x100000 and esp_init_data bin to 0x3FC000.

aeris-ming, 2018-07-01:
I think this device is 512KByte,Flash device data to 0x7C000 works

aeris-ming, 2018-07-03:
@marek26340 I have the same board too.The flash tool said it is 4MB of flash.But if I flash esp-default-init-data not to 0x7C000 "rf_cal[0] !=0x05,is 0xFF" show up then reboot

aeris-ming, 2018-07-10:
@torntrousers  Set SPI Flash configuration to DIO

torntrousers, 2018-07-10:
I can't remember if I tried that. Might have, don't Know. Will give it a go but I've so many modules it might be hard to find that one that wasn't working again.