martin-ger/esp_wifi_repeater #185: Fast led blinking with serial garbage output Esp-01

555learnwithme555, 2018-04-20:
Hello everyone, thank guys for providing this project. I am new here and first time flashing with esp8266 flasher . I tried flashing esp-01 with the bin files and after uploading the firmware i cant communicate or do anything with the module . Only indication if fast led blinking and garbage output. Checked others questions and solutions but nothing helped.. It will be really great if any one could guide me to solve this issue.

martin-ger, 2018-04-21:
Did you read and try this:
Sometimes, especially on old ESP-01s, there is a wrong or non-matching version of "esp_init_data_default.bin" in the flash. If the firmware files from above flash correctly but after reboot you see only garbage on the serial and/or the LED on GPIO2 is flashing rapidly, try to re-initialize this sector: download https://github.com/martin-ger/esp_wifi_repeater/raw/master/firmware/esp_init_data_default_v08_vdd33.bin and flash it to 0x7c000 for 512 kB modules (some ESP-01, Sonoff Switch), 0xfc000 for 1 MB modules (most ESP-01), or 0x3fc000 for 4 MB modules (most ESP-12, Wemos D1). Flashing this sector also

555learnwithme555, 2018-04-21:
thanks for the response ... i have fixed it now by flashing through nodemcu flasher and re flashing the bin file .. Now trying to understand how to use it . Cannot find acess point... I am trying to fix it. thanks alot for the reply

gc9n, 2018-11-09:
I have exactly the same problem, werever i have tried

>  https://github.com/martin-ger/esp_wifi_repeater/raw/master/firmware/esp_init_data_default_v08_vdd33.bin and flash it to 0x7c000 for 512 kB modules (some ESP-01, Sonoff Switch), 0xfc000 for 1 MB modules (most ESP-01), or 0x3fc000 for 4 MB modules (most ESP-12, Wemos D1). Flashing this sector 

 , nothing , keeps flashing and is kinda bricked.

This is my ESP8266 CP2102

https://www.banggood.com/NodeMcu-Lua-WIFI-Internet-Things-Development-Board-Based-ESP8266-CP2102-Wireless-Module-p-1097112.html?rmmds=search&cur_warehouse=USA

gmgunderground, 2018-11-11:
I have the same issue at 115200, i have to connect a scope to see the baudrate

martin-ger, 2018-11-11:
Could you try to go back to https://github.com/martin-ger/esp_wifi_repeater/commit/784c1cf80ec79863b8b1619d59d3c9675114aad4 ?

Does the flashing problem still persist?

gmgunderground, 2018-11-11:
Do you mean flashing configuration at 784c1cf?

martin-ger, 2018-11-11:
No, go back to this version, download the binaries and try to flash these (known to be good).

gmgunderground, 2018-11-11:
Ok, I will try as soon as possible

gmgunderground, 2018-11-11:
no luck, baud is the same 76900, I have only changed 0x2000.bin

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 1328, room 16
tail 0
chksum 0x12
load 0x3ffe8000, len 604, room 8
tail 4
chksum 0x34
csum 0x34

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz

Writing default boot config.
Rom 0 is bad.
Rom 1 is bad.
No good rom available.
user code done

```
[image]

also look the attached file.
With the ESP official tool is the same

martin-ger, 2018-11-11:
Pushed a new 'update' - this version works fine. Please reset the ESP once after flashing.

gmgunderground, 2018-11-11:
Yes @martin-ger the new version is working perfectly, thanks a lot for the support and for the project

closed 2018-11-12 by martin-ger