martin-ger/esp_wifi_repeater #186: how to increase 8 connected stations in AP-mode limit

ulno, 2018-04-21:
I am not sure if this is really the right place to discuss this, but I assume that the people who know the answer to this are reading this here. Would it be better to discuss this on pfalcon's sdk site?

What exactly would have to be done to increase the connection limit to more than 8 stations? Can this be done programmatically here on user-mode side, can it be changed in the sdk, or is it impossible and fixed in a binary blob?

My motivation: I am intending to extend this great project into some kind of smart IoT access point usb-stick for my IoT and Home and Building automation classes and I expect students to connect at least 12-16 devices. However, I don't really need inter-device-communication. All devices only need to see and talk to the accesspoint - not among each other, so I guess buffers for forwarding can be omitted.

ulno, 2018-04-21:
Looking at the esp32 is a great idea - I have some lolin boards back home (will leave europe on Friday, so no real tests before that), I have never tested. Giving the students one esp32 board would not change the overall kit price dramatically - I will definitively take a look. Is there any experience running something like esp_wifi_repeater on the esp32?

martin-ger, 2018-04-21:
Not yet, however, there was a statement from Espressif that they are going to support NAT in the next SDK release.

closed 2018-04-22 by martin-ger