martin-ger/esp_wifi_repeater #190: Access Point with no password

mjf55, 2018-04-27:
Hello, first,  thanks for some great code.  I have it running now and so far, with a simple setup, it is working great.  ( in fact, I am typing this using the repeater )  
I am running on an older NodeMCU ( v0.9) and of course I ran into the fast blinking led issue ( old esp_init_data ).  After I recovered from that, the flashing went OK.  I had some issues setting up the repeater and I wanted to change some settings once I set the parameters to the device.
My environment here is we attach to an open access point, using just the ssid, but no password.  It brings you to a captive portal, where you then authenticate and once that happens, you are connected to the world.
So, I was attached via the serial port and I wanted to reset the password to blank, but was unable to the.  I am at the CMD prompt in the serial communication session, and I enter "set password " ( no password ), I get an error too few parameters.  I could not work past this.  
I got around the issue by erasing / re-flashing and starting from new, not making my typos.  

I hope this makes sense to you.
Minor issue, but wanted you to know.  Thanks again for this tool.

martin-ger, 2018-04-27:
"set ap_open 1"
The ESP will not use a password, whatever this is (you can still use that pw for the "lock" command).

mjf55, 2018-04-28:
Martin, thank you.  I am confused, as to me it looks like "set ap_open 1" will set the local network ( the one this computer attaches to ) as open.  What I was commenting about is the ESP->Router( outside) connection.  I think you call it the STA.  So, when I do show config, it reports back:
STA:   SSID:<router ssid> PW:password   but PW should be blank for this open router.   

Again, I can work around it but just blanking out the Station password ( on 192.168.4.1 ) when ever I change a parameter.

If you do not consider this to be an issue, you can close it ( or I will when you respond. )  

Thanks again.  This is great.

martin-ger, 2018-04-29:
Okay - it always has and shows a pw these - even if is not needed. Simply anter the SSID. If the network is open, any pw is ignored.

mjf55, 2018-04-29:
Thanks.  I just tested that and you are correct.  I put any password in the STA password and I still connected to the open site.  I will close this.

closed 2018-04-29 by mjf55