martin-ger/esp_wifi_repeater #192: Feature Request - Show RSSI when performing a Show Stats

mjf55, 2018-04-29:
When you are in a mesh environment ( I am in a RV park, with many network antennas / repeaters and a single SSID, I think that is called a mesh) it would help me orient or place the ESP-repeater if when I do a **show stats**  the received signal strength is returned.  I would then be able to move the device around to get the best RSSI.
I see in the ESP8266 non-os sdk api_reference, version 2.2, section 3.5.29, there is an api :
**sint8	wifi_station_get_rssi(void)** which should do what I an requesting.  

Is this do-able and does it make sense to you?  thanks

mjf55, 2018-04-29:
So I did it.  I know it was simple, but pretty kool for me.  I added this at line 1081 in user/user_main.c
`os_sprintf(response, "STA Signal Strength %d\r\n", wifi_station_get_rssi());

recompiled and it WORKS.  show stats now shows the Station rssi.  If you like, please include it in your code. If you do, you can move it where you like.

`

martin-ger, 2018-05-01:
Thanks - will add it!