martin-ger/esp_wifi_repeater #195: Inbound traffic to devices connected to repeater not working

gheesung, 2018-05-11:
Thanks for the wonderful work. 

I am trying to setup my ESP sensors within the subnet of repeater (STA IP: 192.168.1.121). All the sensors are in 192.168.4.0/24 subnets and outgoing traffics are ok. They are able to send MQTT messages to my mqtt server (192.168.1.124). 

I have added a static route on my PC, so that all 192.168.4.x traffic can be routed correctly.
`route add 192.168.4.0 mask 255.255.255.0 192.168.121`

From my desktop (192.168.1.79), I am able to ping 192.168.4.1, access to the web console of esp_repeater  and telnet console (port 7777). I have a sensor (192.168.4.5) which has a configuration webpage at port 80. I can only ping the sensor but cannot access port 80 from my desktop. 

Is there anything I have missed out?

martin-ger, 2018-05-11:
In this setting you don't want to use NAT: do a "set net 0" on the ESP.

marek26340, 2018-05-11:
*"set nat 0"

gheesung, 2018-05-11:
Thanks @martin-ger @marek26340 it works!

closed 2018-05-11 by gheesung