martin-ger/esp_wifi_repeater #198: MyAP does not appear after successful flash

corq, 2018-05-30:
I used this a few months ago on a device with no issue. Recently revisited after receiving new Wemos D1's and flashed successfully via esptool. However after reset, myAP never appears. I attempted to add as a possible hidden network but this does not work, either. Any idea what might have changed?

martin-ger, 2018-05-30:
Did you try to write the init sector?

corq, 2018-05-31:
Thank you that seems to have been it, I was sure I had done it earlier in a separate step in preparation but this appears not to have been the case. Much appreciated, and added to my notes!

goosst, 2018-08-28:
I have the same issue: MyAP does not appear after flashing (also no serial communication through arduino IDE seems to occur). I've tried flashing the init file, the other two .bin files in a few variant settings (see screenshot for what I believe should be the correct settings) . When I reboot the wemos D1 after flashing, the led at the wifi chip seems to blink twice and then it's out.

[image]

martin-ger, 2018-08-29:
Try DIO mode.