martin-ger/esp_wifi_repeater #199: connection via rs232 doesnt work

mega-hz, 2018-06-08:
i flashed the two bi files on a nodemcu esp12e but i cant connect via rs232/com.
only garbage is received and set commands/ all commands dont work.
i tryd it on another nodemcu, the same...
i used the esp8266 download tool v3.6.4 on win10

i cant see a new wlan net on my phone ..

can you help?

martin-ger, 2018-06-08:
Did you try DIO mode? Did you try the latest binaries?

mega-hz, 2018-06-09:
yes, dio . the latest binaries? there are only two in this folder, 0x00000.bin and 0x10000.bin.

marek26340, 2018-06-11:
I hope that you followed the instructions "good enough". (0x00000.bin to ESP address 0x00000......)

> Sometimes, especially on old ESP-01s, there is a wrong or non-matching version of "esp_init_data_default.bin" in the flash. If the firmware files from above flash correctly but after reboot you see only garbage on the serial and/or the LED on GPIO2 is flashing rapidly, try to re-initialize this sector: download https://github.com/martin-ger/esp_wifi_repeater/raw/master/firmware/esp_init_data_default_v08_vdd33.bin and flash it to 0x7c000 for 512 kB modules (some ESP-01, Sonoff Switch), 0xfc000 for 1 MB modules (most ESP-01), or 0x3fc000 for 4 MB modules (most ESP-12, Wemos D1).

martin-ger, 2018-06-11:
Thanks, Marek. That's why I asked for the latest binaries. Since last week the binaries try to do this initialization of the init_sector automatically (in user_rf_pre_init() in config_flash.c). If the binary fails and it works after flashing of the init_sector manually, obviously this mechanism isn't as robust as I hoped.

Cheers,
Martin

mega-hz, 2018-06-17:
the one nodemcu was brandnew, but both behave the same...
can you give me help flashing/init those the right way?

mega-hz, 2018-06-18:
i flashed that init file to fc000, then the two bin-files, but stiil the same...
with 74880 Baud i get:

<\r><\n>
ets_main.c <\r><\n>
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)<\r><\n>
<\r><\n>
ets_main.c <\r><\n>
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)<\r><\n>
<\r><\n>
e

martin-ger, 2018-06-18:
Did you check the SHA checksums? How did you download the binaries?

mega-hz, 2018-06-18:
i downloaded the three bin files with "save as"...
how can i check the crc?
i even get this message when i erase the esp!

marek26340, 2018-06-18:
And also, what program are you using to flash the binaries? If esptool, there I had to play around a bit and find the address to get my ESP working.. I was trying 0x3fc00, 0x3fc000, 0x3fc0000.....

martin-ger, 2018-06-18:
Yes, esptool. 0x3fc000 is okay.

mega-hz, 2018-06-18:
ok, i reloaded the complete zip, tried again, the same.
BUT i found the error!
It seems the esp download-tool (win10) doesn't like net-drives, so i copied the binaries to local hdd.
After trieing a bit everything works well!
Thank you for your help!

Just one question:
How can i connect a wlan antenna to the onboard-esp-antenna?

martin-ger, 2018-06-18:
I would not try to connect an additional antenna to the PCB antenna. If this has been done correctly, you will probably destroy the resonance of the antenna. If you want to use an external antenna, use an ESP-07 or an Wemos D1 mini pro - they have a connector.

mega-hz, 2018-06-18:
look https://www.stall.biz/project/esp8266-esp12-mit-externer-wlan-antenne-fuer-wiffi-und-co

mega-hz, 2018-06-18:
great projekt! 
thank you very much!