martin-ger/esp_wifi_repeater #200: examples for use with mqtt suscribe for commands and qio_out

spereg53, 2018-06-09:
can you tellme how to work with mqtt command and output ,i try with node-red and do not recieve nothing on a telnet;the mode publish,work well,thanks

martin-ger, 2018-06-10:
Just publish a command (any line you can also enter on the console) to the command topic (default: "/WiFi/ESPRouter_xxxxxx/command"). You can set the command topic to any value using:
`set mqtt_command_topic <command_topic>`

spereg53, 2018-06-10:
i did this on node-red:
[image: image.png]
in the button i put as payload "scan"but i don´s see nothing on a telenet
session;i can see on node-red the vdd and Uptime as suscriber from publish
of repeater:
[image: image.png]
and please tellme how to control an output on the gpio of esp8266,i am
using esp8266-12,thanks a lot and sorry by my english
ING.SERGIO M.PEREZ GORDILLO
CALLE DE CHAMULA #21
COL.LA LOMITA;C.P.29060
TUXTLA GTZ.CHIAPAS,MEXICO
TEL:019612640373 y
019611214526
CELULAR:9612015537
Email:spereg53@gmail.com
Email:spereg_53@hotmail.com

martin-ger, 2018-06-10:
Sorry, cannot see the images. But this is how to execute commands via mqtt:
Publish a command to: /WiFi/ESPRouter_xxxxxx/command (as you did) and subscribe to /WiFi/ESPRouter_xxxxxx/system/response where you will get the response.

In order to get GPIO output and input running, recompile the sources with those to defines in user_config.h defined:
```
// Define this if you want to get messages about GPIO pin status changes
//#define USER_GPIO_IN   0

// Define this if you want to set an output signal
//#define USER_GPIO_OUT  12
```

spereg53, 2018-06-12:
i recompile the sources and put like suscriber on node-red this topic
/WiFi/ESPRouter_XXXX/switch/0 and nothing  to see ,when i change state on
the pin "0";how is the sintaxis??? thanks
ING.SERGIO M.PEREZ GORDILLO
CALLE DE CHAMULA #21
COL.LA LOMITA;C.P.29060
TUXTLA GTZ.CHIAPAS,MEXICO
TEL:019612640373 y
019611214526
CELULAR:9612015537
Email:spereg53@gmail.com
Email:spereg_53@hotmail.com

martin-ger, 2018-06-13:
The XXXXXX is just a placeholder for the devive Id. Type "show mqtt" on the 
CLI and you will See the real prefix.

spereg53, 2018-07-02:
is a good idea to use the repeater but now i want to use  for monitoring
energy with a pulse to an input logic on pin 2 of esp8266-01 ,so i want to
use the same  repeater and each of esp8266 can give me the pulse from
another sketch; can you helpme????dou speak englis or spanish??
ING.SERGIO M.PEREZ GORDILLO
CALLE DE CHAMULA #21
COL.LA LOMITA;C.P.29060
TUXTLA GTZ.CHIAPAS,MEXICO
TEL:019612640373 y
019611214526
CELULAR:9612015537
Email:spereg53@gmail.com
Email:spereg_53@hotmail.com

martin-ger, 2018-07-03:
If you define USER_GPIO_IN in user_config.h and recompile it, you will get an MQTT message on any state change of GPIO 0 on the toppic _prefix_/GpioIn.

```
// Define this if you want to get messages about GPIO pin status changes
//#define USER_GPIO_IN   0
```

spereg53, 2019-01-07:
i can not get to work gpioin ,i recompile with gpioin on pin 4 and gpioout on pin 5,but i don´t see on my node-red, mqtt subscribe to /prefi/#;can you tell me how, with a example,thanks