martin-ger/esp_wifi_repeater #203: Extremely low speed with one client connected.

yossef12adar, 2018-07-02:
I just flashed the default bin files to my ESP8266 chip, and I set some basic settings through the interface at 192.168.4.1...
the connection work but its extremely slow! nearly two or three minutes to enter google,
I would be glad for some help,
thanks in advance...

marek26340, 2018-07-02:
Have you tried a different browser?

martin-ger, 2018-07-03:
Try to start with a short range. Also try to "ping" the device.

There is no known general throughput problem.

yossef12adar, 2018-07-03:
I tried different browsers and different devices. the repeater is located at a good place and I have 4 bars on the wifi logo...

marek26340, 2018-07-03:
How wide is the bandwidth of your WiFi? I was experimenting one day.. I tried to set BW to 20MHz and the speed of the WiFi repeater increased a bit...

yossef12adar, 2018-07-03:
I relocated the repeater and I managed to achieve 1.92mbits/s (http://www.speedtest.net/result/7441341356)
is that the speed I should expect? anyone succeed to get more speed?

martin-ger, 2018-07-03:
This is at least in the range of what one could expect: many people have reported 4 Mbps and more. Maybe it is an antenna issue. Obviously it is not a performance problem of the ESP...

yossef12adar, 2018-07-03:
OK, thank you all for your responses, closing this issue...

closed 2018-07-03 by yossef12adar