martin-ger/esp_wifi_repeater #205: Multicast is not working.

ibrarahmad, 2018-07-06:


marek26340, 2018-07-06:
Or is it just veeery slow?

ibrarahmad, 2018-07-06:
No, my multicast messages does not reach.

marek26340, 2018-07-06:
Where are the devices that should be receiving the multicast message? They should be all on the same network. The ESP is, by default, running as a NAT router. So if your devices are on your home router's network (for ex. 192.168.171.x), and your ESP's net is (for ex. 192.168.4.x), your multicast messages will only be on the ESP's network (192.168.4.x). They will not reach your home router's network (192.168.171.x). To extend the home router's WiFi and have no NAT in between, you have to disable it in the ESP...

ibrarahmad, 2018-07-06:
There are two issue one is multicast and one is very simple one. I have
this network

 192.168.1.3                   192.168.1.1       192.168.1.2
 192.168.4.1       192.168.4.2
 Laptop1         ------->      HOME_WIFI         ---------->      ESP
<----------       Laptop2

I am unable to ping from 192.168.4.2 to 192.168.1.3 with default
configuration?

marek26340, 2018-07-07:
I can ping any device on my home router's network while I'm connected to my ESP... Is the ESP really connected to your router's WiFi? If it is... maybe try to clear the entire EEPROM of the ESP and reflash it?

ibrarahmad, 2018-07-08:
Yes, ESP is connected to my home route.

CMD>show route

Routing table:

Network              Dest

192.168.4.0/24       ew1

192.168.1.0/24       ew0

127.0.0.0/8          lo0

default              192.168.1.5

marek26340, 2018-07-08:
If your device that is sending the multicast packets is on the ESP's 192.168.4.x network, then it will only reach other devices that are on the same network. It will not reach your home router's network (192.168.1.x). You will have to disable NAT on your ESP, and maybe set some more things up, like the IP addresses, because it will not only start to behave as a bridge (no NAT), but it will also disable DHCP server and client......

martin-ger, 2018-07-08:
The router currently provides no special support for multicasts.

Multicast through a router requires IGMP support on the router. This becomes even more complicated if you use IGMP and NAT combined: https://tools.ietf.org/html/rfc5135 .

closed 2018-10-27 by martin-ger