martin-ger/esp_wifi_repeater #208: ESP8266 restart automatically in AUTOMESH mode

wfs1900, 2018-07-16:
It seems the issue happened here:
if (evt->event_info.connected.bssid[i] != config.bssid[i]) {
                    wrong_bssid = true;
                    os_printf("connect to non configured bssid!");
                    break;
}
but i cannot figure out what is the reason, what is the "config.bssid"?

wfs1900, 2018-07-16:
ESP8266 at mesh layer2 always prints out
connected with ESD_TEST, channel 6
dhcp client start...
connect to ssid ESD_TEST, bssid 24:24:01:d2:08:f2, channel 6
connect to non configured bssid!Saving configuration
state: 5 -> 0 (0)
rm 0
del if0
bcn 0
del if1
usl
disconnect from ssid ESD_TEST, reason 8
=======================================
and ESP8266 at mesh layer1 prints out:
station: 84:f3:eb:4e:2d:c8 join, AID = 1
station: 84:f3:eb:4e:2d:c8 join, AID = 1
station: 84:f3:eb:4e:2d:c8 leave, AID = 1
rm 1
station: 84:f3:eb:4e:2d:c8 leave, AID = 1

martin-ger, 2018-07-16:
So the ESP at mesh layer2 probaly want's to connect to the original root AP (the same as the ESP at mesh layer 1)? I have this problem sometimes, when the two ESP are right side-by-side. Both see the root AP and decide, that it would be the best to connect directly. However, the second then actually connects to the first ESP - and this causes the reset, as this is not what was intended to do...

wfs1900, 2018-07-16:
Yes, my two esp8266 placed side by side, I also thinking that may be the reason. Is it possible the problem is in the comparison process for RSSI，this is my guess.

martin-ger, 2018-07-16:
Perhaps you could try it again with the second node really far away from the root node?

wfs1900, 2018-07-16:
OK, I test the following scenes:
1、Placed first node near the root node and second node in another room , make it near the first node;
==>FAIL
2、Placed both of two nodes in another room and make them a distance apart;
==>FAIL
3、Placed first node near the root node and second node with automesh off, the second node connect with first node successfully;