martin-ger/esp_wifi_repeater #209: reset factory triggers watchdog? other issues too

gr01d, 2018-07-16:
Hi, I tried the original firmware in the repo and whenever I execute 'reset factory' in telnet or serial I am getting the reset failed message:
Reset failed, is this normal?                                                                    
`
Loading default configuration
Saving configuration
Restarting ... 
del if0
bcn 0
del if1
usl
Reset failed
CMD>
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x40100000, len 30256, room 16 
tail 0
chksum 0xf0
load 0x3ffe8000, len 2064, room 8                                               
tail 8                                                                          
chksum 0xf5                                                                     
load 0x3ffe8810, len 5692, room 0                                               
tail 12                                                                         
chksum 0x55                                                                     
csum 0x55         `

How to disable the TKIP since it is not secure, I managed to enable WPA2 only but not to be able to change channel and disable TKIP (esp_open_sdk). The adc_read I cannot read any value that makes sense, I am using an external circuit 0-1V only. At last, how do I set the sntp time manually, I cannot find it in the code. A method that I am planning is to use a javascript in the web.h to get the time in client connected to the ESP and set the time from there.                                                             
I cannot unlock the device after checking the Lock Device option, I tried the STA password but no avail, can I just hardcode it where?

What is also the license of the program?

Summary:
- how to change wifi channel (randomly too)?
- how to disable TKIP?
- how to read analog TOUT if connected to external circuit?
- how to set time (sntp) manually (if no internet or DS3231 RTC)?
- what is the license of the program?
- lock device won't accept my STA password as unlock code?

Thank you for this software

gr01d, 2018-07-16:
is this program KRACK vulnerable? any way to fix it?

martin-ger, 2018-07-16:
Let me try to answer some of your questions quickly:
- All firmware published after 17/Oct/2017 have been built with the patched version of the SDK 2.1.0 from Espressif that mitigates the KRACK (https://www.krackattacks.com/ ) attack. (see known issues)
- The Wifi channel always has use the same channel as the uplink network
- No idea how to disable TKIP - is there any API in the espressif SDK?
- analog TOUT is controlled by the 107th byte of esp_init_data_default.bin (0 - 127 byte) is “vdd33_const“. It must be set to 0xFF i.e. 255 to read system voltage i.e. voltage on VDD pin of ESP8266.  Init data is in config_flash.c (change this byte if you want to read the ADC)
- lock "should" work with the uplink password, however you can choose your own by setting "lock <passwd>

martin-ger, 2018-07-16:
licence? My code is free - several contributions have their own headers. Basically all open, as long as you mention their names...

gr01d, 2018-07-18:
Thanks for replying, I have yet to test for the krack vulnerability, for the TKIP I did not find anyway to disable it not even in the user_interface.h in the sdk. For the adc_read on external circuit I found another implementation of it that just works but still unable to test fully. For the ntp question I just found a library (zarya and richard's) and managed to get it to work with a minor issue in encoding year.

 But the lock password it does not work for me, I specified a sta ssid password on user_config.h and tried to set the config->locked			= 0 to config_flash.c but the unlock code is not accepted.

issuing a reset factory in serial causes a wdt reboot: 2 and config reset failed but it still appears to reset the config..

martin-ger, 2018-07-18:
The lock issue:

Try to enter at the cli:
`lock 123`
does it lock the config?
If so, try:
`unlock 123`
For me, it unlocks...

The reset issue:
It seems to okay: I call system_restart() - the docs says: "The ESP8266 will not restart immediately. Please do not call other functions after calling this API."
I will try an infinite waiting loop after restart...

gr01d, 2018-07-19:
For the lock issue, if I used the http server like page and tick the checkbox and the Lock button it locks but the code is \0? NULL? manually setting the lock in telnet works fine. 
Also, after reloading the http page for 5 + times the serial output it E:M 2400 and the wifi nat no longer works, this may be because of my custom version of your program. But I think it is related to a memory full?

gr01d, 2018-07-22:
E:M 2400 means that the memory (RAM) is at 2400 below, to fix it I just added the os_free in the page_buf codes. I am closing this issue now, thank you for your help.

martin-ger, 2018-07-22:
Where did you add the os_frees. Would like to fix that in mainstream if possible.

gr01d, 2018-07-23:
I added os_free in the static void ICACHE_FLASH_ATTR **web_config_client_connected_cb**(void *arg)
before the page_buf = NULL;, I added  os_free(page_buf);

.
.
.
    espconn_send(pespconn, page_buf, os_strlen(page_buf));
    **os_free(page_buf); //ADD FIX**
    page_buf = NULL;
.

to monitor the mem usage I added the: os_printf( "Free mem: %d\r\n", system_get_free_heap_size());
to a timer and watched the serial output while I reaload the config page hundresds of times.

martin-ger, 2018-07-23:
Thank you for your fix and the bug reports.

I now also fixed the lock-issue in the Web-Interface.

gr01d, 2018-07-28:
In the lock, I think you may need to change it to config.password for the STA password, it is set on AP password.

		    os_memcpy(config.lock_password, **config.ap_password**, sizeof(config.lock_password));

TO:
		    os_memcpy(config.lock_password, **config.password**, sizeof(config.lock_password));

I found a bug? the sleep function, after waking up it just not work anymore. Here is the serial output upon executing sleep:
```
CMD>sleep
Going to deep sleep for 10s
CMD>del if0
bcn 0
del if1
usl
��sdd��|�$�|p
               �l�
                   c|����{�cd�#��'o�dog���
                                            bx
                                               dc$sd�r��g�b�g#x�

```

martin-ger, 2018-07-30:
Fixed the ap_password thing...

martin-ger, 2018-07-30:
For me "sleep" works well on a Wemos D1 mini with GPIO16 connected to RST. It fails on a board like this https://www.ebay.de/i/172720108707?chn=ps with a LED on GPIO16.

gr01d, 2018-08-01:
sleep does work, I just forgot to connect GPIO16 (D0) to RST

closed 2018-08-01 by gr01d