martin-ger/esp_wifi_repeater #212: AutoMesh to drive a massive network

jadolg, 2018-07-22:
This is more a question than an issue, but here it goes:
Can I run a massive mesh network only with this? Let's say I'm interested in serving multiple websites in a community. I'm talking about 1000 devices mesh plus client devices. Can it be done using AutoMesh mode? What limitations will I have? (I'm assuming it will be a slow network)

martin-ger, 2018-07-22:
Don't think so: limitations are:
- one ESP cannot serve more than 8 STAs
- overall bandwidth and airtime (all must run on the same channel, each packet has to be transmitted several times, may collide on air)
- because of that packet: loss rate

BTW, I am not aware of ANY mesh-like network that can handle that.

A typical szenario for my "automesh" would be a handful of IoT-devices sending from time to time.

jadolg, 2018-07-22:
I have used your software before to share my internet connection and the only problem till now is that NAT table gets full every few minutes and then everything is slower until I reset. Does this problem persist in AutoMesh mode?

jadolg, 2018-07-22:
And btw, modem speed is still acceptable at some point. Would it be achievable?

martin-ger, 2018-07-22:
Automesh uses the same NAT table meachnism as the "normal" repeater mode. So, yes, this will persist. One might increase the NAT table a little bit by recompiling LWIP. Still I don't think, that hundreds of nodes will work - without further experiment, I would expect congestion in the order of several tens...

closed 2020-12-08 by jadolg