martin-ger/esp_wifi_repeater #216: re-direct web server via Wifi extender????

hwarang098tkd, 2018-08-03:
So Hi all!!!
I have a rasp with openhab server. IP 192.168.1.43
my desktop on IP192.168.1.3
I have also a Wi-Fi extender(esp_wifi_repeater )  192.168.1.34
And 3 clients(sonoffs touch with tasmota ) connected on extender.
So my question is: How can I see the websevers on these 3 clients on my Desktop in ip 192.168.1.3.
Of course I can see them through Wi-Fi extender if I connect with my phone.
thnx in advance

martin-ger, 2018-08-03:
As the esp_wifi_repeater uses NAT, you need portmaps:

Given that the three sonoffs are connected and have the IPs 192.168.4.[2-4]. Do a "save dhcp" to fix the IP mapping first. Now enter on the CLI:
```
portmap add TCP 80 192.168.4.2 80
portmap add TCP 81 192.168.4.2 80
portmap add TCP 82 192.168.4.2 80
save
```
Now you can access the three Web-Interface at 192.168.1.34:80, 192.168.1.34:81,and 192.168.1.34:82.

Note that MQTT from the sonoff to an Mosquitto on the rasp doesn't need this, as the connections are set up from the sonoffs to rasp, so NAT works automatically.

hwarang098tkd, 2018-08-03:
works fine .thnx dude

closed 2018-08-03 by hwarang098tkd