martin-ger/esp_wifi_repeater #218: Disconnect from AP after x minutes when no clients connected

projectb-temp, 2018-08-14:
Hi there Martin,

Lately I've been testing your NAT router (in combination with an AVM FritzBox) and it runs quite nicely.

In order to reduce power consumption I configured the FritzBox to power down its WiFi if there are no active clients connected for some period of time. This runs flawlessly so far.

Now, when using the NAT router the FritzBox WiFi won't power down as the NAT router is always connected to the FritzBox network - even if no "real" clients are connected to either the NAT router or the FritzBox.

How can I tell the NAT router to automatically disconnect from the FritzBox AP (and stay disconnected for a couple if minutes) so the FritzBox can shut down WiFi?

It's perfectly fine if the Nat router tries to reconnect to the FritzBox AP every now and then to re-establish a connection as soon as the FritzBox WiFi is enabled again.

An option on the NAT router to "disconnect after x minutes with no client connected and try a reconnect every y minutes" would be great.

What do you think about this? Or is it even possible with the actual firmware?

Thank you  for your response.
Best regards.

projectb-temp, 2018-09-07:
@martin-ger
Dear Martin,
as I got no response from you I'm pushing this request once.

Could you please be so kind as to look over my issue.
Thank you and best regards.

gr01d, 2018-09-14:
You mean if there are no client connected to the esp8266 it should go to sleep for a defined time then this will tirgger the FritzBox WiFi to shutdown wifi too? If you have an experience with C programming you can implement it yourself, I will try to implement it as well.

projectb-temp, 2018-09-15:
Yes, I meant something like this.

The NAT router doesn't neccesarily need to be in a sleep mode when the last client disconnects. A simple disconnect from the FritzBox WiFi for a couple of time is sufficient to trigger the powerdown of the FritzBox WiFi.

Assuming that:
- At least one client is connected to the NAT router
- The NAT router is connected to the FritzBox WiFi

It could work like this:
1) The last client disconnects from the NAT router
2) The NAT router waits some (user defined) time before disconnecting from the FritzBox WiFi if no client has been connected since the last one has disconnected
3) The NAT router stays disconnected from the FritzBox WiFi for a (user defined) period of time, so that the FritzBox can decide to power down it's WiFi (in case the NAT router was the last client connected to the FritzBox)
4) The NAT router periodically tries to reconnect to the FritzBox WiFi within a (user definable) time
5) The NAT router can re-connect to the FritzBox WiFi as soon as it get's available again and can wait for clients a (user definable) time after the re-connect before disconnecting from the FritzBox WiFi again
6) Start over @ 4) until a client connects and then start over @ 1) again

This would mean there are 4 user definable times that would have to be added to the setup:
- Timeout before disconnecting from the FritzBox WiFi when no client ist connected
- Time to stay disconnected after the NAT router has disconnected from the FritzBox WiFi because of no client connected
- Time to wait between periodical tries to re-connect to the FritzBox WiFi
- Timeout to stay online after a re-connect to the FritzBox WiFi so that clients have time to connect to the NAT router

It would be really great, if you could try and implement this into the current solution, @gr01d !

Thank you and best regards

gr01d, 2018-10-03:
Hi, the ssid_watchdog and ap_watchdog has may be the function that you want. Case 1 the ssid_watchdog, the NAT router or your esp device cannot connect to the specified ssid it will reset itself, if you know C programming you can set this to make the esp device sleep for your defined time, case 2 the ap_watchdog, the esp device will reset, restart if there are no devices connected to it, again you can probably change the code a little to make it sleep for predefined time. I would like to implement this but I am very busy.