martin-ger/esp_wifi_repeater #222: No booting with ENC28J60

vfperri88, 2018-09-09:
Hi, could anyone make it boot whith the ENC28J60 connected? I follow the instructions to the connection but my ESP don't boot any more unless I disconnected the GPIO15 and connected it to GND. My connection schema is the following and I'm using the transistor for decoupling GPIO15.

NodeMCU/Wemos  ESP8266      ENC28J60

        D6     GPIO12 <---> MISO
        D7     GPIO13 <---> MOSI
        D5     GPIO14 <---> SCLK
        D8     GPIO15 <---> CS
        D1     GPIO5  <---> INT
               Q3/V33 <---> 3.3V
               GND    <---> GND

Thanks!

Victor

martin-ger, 2018-09-11:
If you try to pull GPIO15 low with the decoupling transistor directly at the pin - let's say with additional 3,3kOhm?

vfperri88, 2018-09-22:
> If you try to pull GPIO15 low with the decoupling transistor directly at the pin - let's say with additional 3,3kOhm?

Thank you very much @martin-ger. I could solve the problem booting putting an additional pull down resistor.

closed 2018-09-22 by vfperri88