martin-ger/esp_wifi_repeater #224: Setup page incomplete

tedhale, 2018-09-14:
I have downloaded the latest BIN files to a generic ESP8266. When I connect to the AP to do setup, I get an incomplete page, preventing me from completing setup. Verified that power is not the cause of the problem.
image below.
[image]

tedhale, 2018-09-14:
Problem occurs using Windows 10 and Chrome. However, Chrome on my android phone works correctly.

gr01d, 2018-09-15:
have you tried reloading the page multiple times? it happens to me but reloading the page is the fix

projectb-temp, 2018-11-29:
Eventually you could do a [Ctrl]+[F5] to force a cache-refreshed reloading of the page and see if that helps.
Just be sure not to spam the device with reloads.