martin-ger/esp_wifi_repeater #228: Feature request

blucmal, 2018-09-23:
Wondering if it would be possible to have the unit sleep and wake on a schedule to save batter power?? I use it for iot devices and they only need web access every 6hrs..

herrxyz, 2019-01-12:
timed sleep should be possible by using ntp, to wake up an esp there is a need for a connection between D0 and RST. Unfortunately, this connection blocks firmware-upload, which means you shouldn't solder it.
for many users, this could be a problem, if they are not aware of that

martin-ger, 2019-01-12:
There is no other way to send the ESP to deep sleep. Using NTP implies, that the ESP is active.

herrxyz, 2019-01-12:
it should be possible to use ntp to calculate deep sleep timer

herrxyz, 2019-01-12:
example pseudocode: shutdown at 10am, wakeup at 11am
```
while true:
  if (10am - now < 0)
    deepsleep(calc(11am-now))
```

martin-ger, 2019-01-12:
But still you need to connect GPIO16 and RST. Also deepsleep only works for max 71 min.

herrxyz, 2019-01-14:
didn't know about that deepsleep limit until now ;-( thanks for that info!
[here](https://thingpulse.com/max-deep-sleep-for-esp8266/) it seems possible to extend it to 3-3,5 hours, but still not a great solution.
to work around it, it could sleep for 3 hours, then reboot and recalculate next sleep cycle
might be usefull to save a bit of battery