martin-ger/esp_wifi_repeater #232: Not in sync with martin-ger/esp-open-lwip library

xxxajk, 2018-10-08:
Simple removal of the polling to the enc28j60 in the code fixes the issue.
I've been testing the latest library, which is ISR driven, and it works very well.

martin-ger, 2018-10-08:
Don't exaclty understand the issue. You suggest a change in the repeaters code or in the lwip's?
Any hint appreciated.

xxxajk, 2018-10-08:
This app is not in sync with your other library, which uses ISR instead of polling.
Briefly: 
The file include/lwip/netif/espenc.h is out of date
All references to enc28j60_poll() and SIG_ENC get removed
The timer to poll gets removed.

Then it will be in sync and build without issue with the latest library from your own repository set.

xxxajk, 2018-10-08:
P.S. I'm working on DHCP server for the enc28j60 right now too, as it is needed for a project. 

Off-topic: I've also written a very lightweight network stack since lwip tends to be a bit convoluted, so, yeah, I know what I'm doing and talking about ;-)

martin-ger, 2018-10-08:
Thanks! Compiles fine - will test it in the evening.

xxxajk, 2018-10-08:
Yeah, I would have done a pull request, but I've changed quite a lot in the library, etc, probabbly too much for anything to make sense. I'll probabbly make an independent fork, which just has everything put together under a top-level directory (including the espressif tools). If you like that idea, let me know and I will totally make a fork with credit pointing back here, but it won't be the usual git style fork, unfortunately, because the way I can lay everything out makes the headers/etc always up to date.

martin-ger, 2018-10-08:
Would be great - think I will be able to include it into my project.

xxxajk, 2018-10-08:
As soon as I find out why packets are leaking to the wrong interface, it
will be good. Right now packets leak to the DHCP server on the wifi
interface... I don't think that they make it to mine, which should not
happen if it is bound to an 8nterface. Could be an lwip bug, not sure yet.

xxxajk, 2018-10-09:
It seems lwip has a very nasty routing bug when it comes to broadcasts to listening sockets and receivers... So far I have a 50% bug fix. 
Packets are not correctly directed with inbound or outbound UDP, probabbly fails on TCP too. 
It will probabbly be a 4 line fix, since inbound was just a 2 line addition.

What can happen is that you could spam the enc port with fast DHCP requests and they would end up going to wifi's dhcp server, and the replies end up going there too... That could knock wifi out with an indirect ARP poisoning!
So far I have fixed the inbound problem. Once I fix the outbound issue, I'll do a virgin pull and give a pull request to fix it.

xxxajk, 2018-10-14:
Did a pull request for the library. 
See https://github.com/martin-ger/esp-open-lwip/pull/6
Will have one for this repo shortly.

xxxajk, 2018-10-14:
Pull request submitted that has all the new goodies. Enjoy!

closed 2018-10-16 by xxxajk

xxxajk, 2018-10-16:
One final question, perhaps you know, perhaps you do not...
Take a look of setup and results:

```
CMD>set eth_enable 1
eth enabled
CMD>set eth_ip 192.168.5.1
ETH IP address set to 192.168.5.1
CMD>set eth_netmask 255.255.255.0
ETH IP netmask set to 255.255.255.0
CMD>set eth_dhcpd 1
eth_dhcpd enabled
CMD>save config
Saving configuration
Config saved
CMD>reset

Restarting ...
del if0
bcn 0
del if1
usl

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x40100000, len 1328, room 16
tail 0
chksum 0x12
load 0x3ffe8000, len 604, room 8
tail 4
chksum 0x34
csum 0x34

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  40 MHz

Booting rom 0.

(deleted garbage)
WiFi Repeater V2.1.0 starting

running rom 0
Config found and loaded
Starting Console TCP Server on port 7777
Starting Web Config Server on port 80
Starting enc
Starting enc DHCPd
mode : sta(dc:4f:22:26:98:4d) + softAP(de:4f:22:26:98:4d)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
(plug in network cable)
E:M 61048 <-- What
E:M 65616 <--      are
E:M 24704 <--          these
E:M 24704 <--                telling
E:M 24704 <--                        me?

```
As it says, what's the E:M values? I can't locate them as an output anywhere... some sort of error?

RibeiroSantos, 2018-10-16:
I don't know but hope this could help:

https://www.esp8266.com/viewtopic.php?f=24&t=12868
https://github.com/micropython/micropython/issues/1971

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Sem
vírus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

à(s) 23:30:

xxxajk, 2018-10-16:
Thank you very much! Appreciate it! I won't pester you any more until I get this going from the lib again.

martin-ger, 2018-10-17:
These are out of memory messages coming from somewhere deep inside the memory management. Typically, when lwip buffer allocation fails. I get these sometimes from the WiFi driver in overload situations.

xxxajk, 2018-10-17:
E:M 65616
Yeah, but 65K isn't enough RAM? Something else is wrong. 
Most complaints/reports show much lower values.
I'll have to try a newer SDK, which has given better results even under Arduino.

xxxajk, 2018-10-17:
Arduino uses 1.20.0-26-gb404fb9-2...
whereas the repository pointed to by this project uses 2.1.0-18-g61248df
Quite dated, not that it is normally a problem.
There are a ton of bug fixes and security issues fixed. I had massive WiFi issues before on Arduino, which have gone away with the latest update.

RibeiroSantos, 2018-10-17:
Sorry just suggesting links but this is my best :D

https://bbs.espressif.com/viewtopic.php?t=2756

martin-ger, 2018-10-17:
Just guessing (I am on a business trip now): for some reason the driver tries to allocate an invalid packet length. 0xffff+some header size...

xxxajk, 2018-10-17:
The problem with this is that after that happens all network activity is
dead. The console is still alive though.

xxxajk, 2018-10-17:
One thing that helps a little bit, is if I set NAPT to dynamic, then it tends to hold off long enough to send a DHCP packet over the wired ethernet, and after that, no activity, except memory problems.

martin-ger, 2018-10-17:
NAPT dynamic means less statically preallocated memory. My guess would still be wrong: packet lenghs... Could you instrument buffer alloc in the driver?

xxxajk, 2018-10-17:
I'll check on leak debug and the os malloc in a bit, and see what I can
report.

martin-ger, 2018-10-22:
For me it seems to work, as long as debug log in the enc-driver is on. If it it off, it seems that incoming packets are not received...

martin-ger, 2018-10-23:
Okay, did some further investigations:
- When logging is off for some reason the enc-driver reads in enc28j60_handle_packets() with readBuf() a NextPacketPtr of 0xffff. This is wrong and the following code usually reads a random packet size and tries to allocate a much to large buffer, resulting in the E:Mxxxx error.
- After that the driver is out of sync with the reveive queue and continues to read from 0xffff wrong packets.

martin-ger, 2018-10-23:
Further result:
- If I remove all logs but the one in line 80 of readBuf() (log1("readBuf()");) it runs pretty reliable.

Guess it has to something with SPI timing...

martin-ger, 2018-10-23:
Think i got em:
A "while(spi_busy(HSPI));" before the SPI transaction seems to get it right...

xxxajk, 2018-10-23:
A timing problem was one of the things I was also thinking of. Good find!
I'll try it here immediately.

xxxajk, 2018-10-24:
Still testing, this is looking good!

xxxajk, 2018-10-24:
Test 1 flood ping test parameters: 1000 packets to the wifi IP address from the ENC connection. No WiFi peers connected. 80MHz

56 data bytes PASS
500 data bytes PASS
750 data bytes PASS
800 data bytes FAIL: 27% packet loss, device complains E:M or reboots. Needs powercycle to recover

Test 2 flood ping test parameters: 1000 packets to the wifi peer from the ENC connection. 80MHz
56 data bytes PASS
500 data bytes PASS
750 data bytes PASS
800 data bytes FAIL: 95% packet loss, device complains E:M or reboots. Needs powercycle to recover

Test 3 flood ping test parameters: 1000 packets to the wifi IP address from the ENC connection. No WiFi peers connected. 160MHz
800 data bytes PASS
900 data bytes PASS
1000 databytes FAIL: 38% packet loss, device complains E:M or reboots. powercycle required to restart
1200 data bytes PASS: 19-30% packet loss 
1300 data bytes PASS 19-30% packet loss 
1400 data bytes PASS 29-48% packet loss 
1450 data bytes PASS 99% packet loss
1454 data bytes FAIL , 61% packet loss, device complains E:M or reboots. powercycle required to restart
 
Test 4 flood ping test parameters: 1000 packets to the wifi peer from the ENC connection. 160MHz
800 data bytes PASS
1000 data bytes PASS
1200 data bytes PASS: 19-30% packet loss 
1300 data bytes PASS 19-30% packet loss 
1400 data bytes PASS 29-48% packet loss 
1450 data bytes PASS 99% packet loss
1454 data bytes FAIL , 61% packet loss, device complains E:M or reboots. powercycle required to restart

Sometimes good, sometimes not on some of the tests. It depends if the tablet I have connected is trying to "phone home" so that could be the reason for some of these crashes.

Still, this is working very well!

martin-ger, 2018-10-24:
Think, the E:M (with sizes of around 1.5KB) happens due to general memory shortage, this also happens when the wifi is overloaded.

The power cycle is probably needed as we do not have/use the HW-reset. Perhaps we could enhance the driver to use it.

The 1450-1454 byte border is certainly due to the 1500 MTU-size. Packet + Payload need fragmentation - lwip can't handle that.

xxxajk, 2018-10-24:
1454 should *just* fit a few, yes, lwip has no fragment handling. 
I was trying to see if it could actually be crashed.
160MHz is probably the way to go, seems to keep up better.

martin-ger, 2018-10-24:
Just added a HW-reset to the code in user_main.c (but not in default config):
Connect GPIO 4 to RESET of ENC and in user_config: 
- FACTORY_RESET_PIN to something beyond 16
- HAVE_ENC28J60 1
- ENC28J60_HW_RESET 4

If it works, should it go into the driver?

xxxajk, 2018-10-24:
I have mine connected to the esp's reset line... Normally the soft reset should be good enough, but might not be.

xxxajk, 2018-10-27:
That's where I'd put it.

xxxajk, 2018-11-01:
figured it out... turned off NAT, added another default route to the eth, done!
Now to really beat it to death and see what explodes!

xxxajk, 2018-11-01:
Seems after one full-sized packet, that no more packets are allowed to pass.
Have to break out wireshark, it seems...

xxxajk, 2018-11-01:
Wireshark just said there was a missing packet (TCP Previous segment not captured)
 sooo... turned on debug and found this:

```
[lwip/netif/espenc.c:enc28j60_link_output:133] output, tot_len: 1514
[lwip/netif/espenc.c:enc28j60_link_output:135] link is up: 1
[lwip/netif/espenc.c:enc28j60_link_output:136] pktcnt: 0
[lwip/netif/espenc.c:enc28j60_link_output:156] before transmission: 00
[lwip/netif/espenc.c:enc28j60_link_output:169] transmission failed (132 - 02)
[lwip/netif/espenc.c:enc28j60_link_output:133] output, tot_len: 1514
[lwip/netif/espenc.c:enc28j60_link_output:135] link is up: 1
[lwip/netif/espenc.c:enc28j60_link_output:136] pktcnt: 0
[lwip/netif/espenc.c:enc28j60_link_output:156] before transmission: 00
[lwip/netif/espenc.c:enc28j60_link_output:169] transmission failed (133 - 0a)
```
This is going to be fun... or not...

xxxajk, 2018-11-02:
Basically, this IP stack is lacking MTU discovery.
It needs to do an ICMP reply (Type 3, Code 4) with the message containing its MTU.
What is strange is that the packet with this size made it into the WiFi side.

xxxajk, 2018-11-02:
https://github.com/martin-ger/esp-open-lwip/blob/sdk-1.5.0/lwip/core/ipv4/ip.c#L1024
This needs to check MTU and reply if too large :-) It doesn't.

Still unsure how the 1514 size made it in, because the MTU on the source is 1500, and reassembly is indeed off.
Shall I do this or???

xxxajk, 2018-11-03:
Designed in MTU discovery across interfaces, also found 1 bug in the enc driver, and fixed that. 

Last thing to figure out is why heavy traffic from/to tablet kills the ip stack... 
Both wifi and eth gateways ping from eth after WiFi disappears.
Finally, memory "leaks" until filled and wdt reset occurs.

xxxajk, 2018-11-03:
I'll push and make a pull request shortly.