martin-ger/esp_wifi_repeater #233: Can't reach IPs connected to Repeater

danieloSD, 2018-10-08:
My main network is DHCP is on 10.0.0.x and the repeater sets it's own DHCP to 192.168.4.x .  I can not ping a 192 address from the 10 network.  Isn't there a way that the repeated doesn't use it's own DHCP so that the machines connected to the repeater get a 10. address ?

martin-ger, 2018-10-08:
No, the "repeater" is actually a NAT router and the ESP's network is a separate IP network with it's own addresses (distributed by the DHCP server).

Theoretically it would be possible to switch off the NAT and use plain IP routing for a full two-way connectivity between the AP and the STA network. However, this requires static routing entires in all clients - probably not what you want to configure.

If you need single services accessible in the ESP's network use portmap entries instead.

xxxajk, 2018-10-08:
You can always set the route manually, pointing 192.168.4/24 to the IP given by the DHCP server on the main gateway router. That is what worked for me with a static IP to MAC address in the router box.

I tried to do something similar with the enc28j60 driver, by setting the IP to the same subnet as WiFi, and that causes the program to die in a watchdog timeout... Of course this would be a bridge, not NAT, so I was just playing and seeing what I could do to make it fall apart.