martin-ger/esp_wifi_repeater #234: SOS! SOS! emergency!emergency! Why can‘t i modify the ap_mac???

unix1024, 2018-10-12:
I use the command "set ap_mac xx:xx:xx:xx:xx:xx" to modify the esp8266's mac.   And the tool display i have modify the mac.   
[image]

But!!!But!!!  when i upload the program to esp8266, it still display the old mac!!!
[image]

moreover!!   I use my computer to connect the esp8266, and catch the packet ,   the destination mac still the old mac 
[image]

SOS! Help!

unix1024, 2018-10-12:
please help me , thank you ! God bless you , Dear.

My we-chat: 17801141216
My phone : 17801141216
My e-mail: unix1024@163.coom

martin-ger, 2018-10-16:
Change is only temporary for the wifi_repeater. Other programs will still read the original MAC from ROM.

unix1024, 2018-10-22:
Oh，thank you very much for your help. I have found the question. I don’t execute the command to save the config. Oh，my god. I am so idiot！

closed 2018-10-22 by martin-ger