martin-ger/esp_wifi_repeater #235: Feature Request mqqt IO

rebe0909, 2018-10-13:
Hello.
There ist already mqtt on Board and there are some IO Pins free.
So I am wondering if it would easy be possible to integrate an mqtt IO?

e.g. to be configured over the telnet console?
Thanks in advance!
Kind Regards 
Reinhard

martin-ger, 2018-10-15:
If you enable USER_GPIO_IN and USER_GPIO_OUT in user_config.h and recompile you can at least use one digital input and one output pin. 

You can control the output PIN via an MQTT topic (define it with "set mqtt_gpio_out_topic xxx") and get state changes of the GPIO_IN pin via MQTT "<prefix>/GpioIn" topic .

Not much of docs right now, but you can see the details in the code of user_main (look for "#ifdef USER_GPIO").

bchabrier, 2019-04-10:
With #293 all GPIO state changes are published via MQTT `/Gpio` topic.