martin-ger/esp_wifi_repeater #244: Detected overlap at address: 0x2000

haaslukas, 2018-11-15:
I just wanted to flash a Wemos D1 mini by using this command:
`esptool.exe --port COM4 write_flash -fs 4MB -ff 80m -fm dio 0x00000 0x00000.bin 0x02000 0x02000.bin`
which gives me the error 
`Detected overlap at address: 0x2000`

When I change from 0x02000 to 0x10000, the esptool flashes the Wemos, but I'm neither seeing the AP's SSID nor anything on the serial interface (event after reboot and reset)

`esptool.exe --port COM4 write_flash -fs 4MB -ff 80m -fm dio 0x00000 0x00000.bin 0x10000 0x02000.bin`
gives me

`esptool.py v2.5.1`
`Serial port COM4`
`Connecting....`
`Detecting chip type... ESP8266`
`Chip is ESP8266EX`
`Features: WiFi`
`MAC: dc:4f:22:1d:8c:64`
`Uploading stub...`
`Running stub...`
`Stub running...`
`Configuring flash size...`
`Compressed 56956 bytes to 13843...`
`Wrote 56956 bytes (13843 compressed) at 0x00000000 in 1.2 seconds (effective 372.0 kbit/s)...`
`Hash of data verified.`
`Compressed 56972 bytes to 13817...`
`Wrote 56972 bytes (13817 compressed) at 0x00010000 in 1.2 seconds (effective 374.2 kbit/s)...`
`Hash of data verified.`

Any idea what to test next?

martin-ger, 2018-11-15:
How long is your 0x00000.bin? It should fit before 0x02000.

haaslukas, 2018-11-16:
stupid me. I did a right click on the files in the esp_wifi_repeater/firmware/ folder and saved them without actually checking the .bin contents on their plausibility or their length. it turns out that I actually downloaded the html files... with the right files everything works of course.

closed 2018-11-16 by haaslukas