martin-ger/esp_wifi_repeater #245: 'Basic Web Config Interface' refuses to open

manj9501, 2018-11-17:
At first boot after flashing the latest firmware (with OTA support) on either my generic esp8266 or NodeMCU (_0x00000.bin_ at _0x00000_ and _0x02000.bin_ at _0x02000_, respectively) using ESP8266 Download Tool, an open access point _MyAP_ is created. However once I connect to this AP on my Windows 7 PC, it shows 'Unidentified network with no network access' as the network status and the _Basic Web Config Interface_ (http;//192.168.4.1) refuses to show up in the browser. Interestingly, the CLI is still accessible, though I don't know the extent to which it's actually functional. I've also tried flashing the _esp_init_data_default_v08_vdd33.bin_ at _0xFC000_ (in my generic esp8266), to no avail. 
Am attaching the log I obtained in the serial monitor after I gave the _reset_ command in the CLI, and connected my PC to _MyAP_.

Note: An old version of the firmware, which doesn't have OTA support, works without any hassles on both the devices.
[image]
[image]

tgondek, 2018-11-18:
I have the same issue

martin-ger, 2018-11-19:
Do you use the latest binaries? Is it possible to display the Web-Interface on e.g. an Android Smartphone?

manj9501, 2018-11-19:
**Yes**, the latest binaries are being used and as was mentioned previously, the problem seems to occur only while using the latest binaries. Secondly, when trying to connect to _MyAP_ on an Android smartphone (a Samsung specifically), a _Connecting..._ message is shown for way too long, with the smartphone eventually showing a message saying _Failed to obtain IP address._ (Screenshot attached), before beginning to connect to other available APs **OR** it does successfully connect, once in a while, to _MyAP_ with the Web-interface still not loading. The former scenario is the dominant one, happening every 9 out of 10 times an attempt to connect to _MyAP_ is initiated.
[image]

schup, 2018-11-21:
I have the same issue on a ESP01 with direct flashing.

Last working commit is 2a5350a6124b50ac73ae6140575756fafd3d60ad - Merge pull request #240 from xxxajk/enc_polling 

Seems like 830c8b69943724f9f98a9e3a5f49ebda0677c5fc - Run enc28j60 reads in seperate task introduced the problem.

ljrob61, 2018-11-21:
Connect to MyAP then go to your web browser and enter the 192.168.4.1 while the browser is searching for the page. Press the reset button or power down then power up the ESP8266. After a while the page should appear. This also happens if you do a "reset factory" you lose the web page. I have done this several times and it has always worked.
Hope this helps find the problem.

martin-ger, 2018-11-23:
If you experience these problems,please try to go back to https://github.com/martin-ger/esp_wifi_repeater/commit/2a5350a6124b50ac73ae6140575756fafd3d60ad and uses these binaries. They were compiled with the 2.1 SDK. I will further investigate...

martin-ger, 2018-11-26:
Seems to be a problem, when compiling with SDK 2.2. Have gone back to 2.1 - for me it works again.

tgondek, 2018-11-26:
Good news.  Are there new binaries oline for downloading?

Tom

closed 2018-11-28 by martin-ger