martin-ger/esp_wifi_repeater #246: Connection Drops after 2 minutes an then won't connect

DetainedDeveloper, 2018-11-23:
I have to unplug and replug the nodemcu to work again.
Though it works only for two minutes then it stops working as my devices show "Obtaining IP Address" forever.

Alejandro1407, 2018-11-26:
Hello I Had same problem with ESP8266-01, and the only way that i finded to solve this is use a previous version of the project specifically this: https://www.instructables.com/id/POWERFUL-Wi-Fi-REPEATER-NODE-MCU/ following that tutorial i was able to solve that problem but like disadvantage you can't configure repeater using interface web

martin-ger, 2018-11-26:
Seems to be a problem, when compiling with SDK 2.2. Have gone back to 2.1 - for me it works again.