martin-ger/esp_wifi_repeater #250: Add DNS Please

wieb18, 2018-12-09:
When my ESP8266 connected to router or wifi repeater, it automatically got DNS setting.  So no problem when using host name eg. "www.google.com" because it will return the host IP address automatically.

But unfortunately, this esp_wifi_repeater **forgot** included automatic DNS setting. Host IP address always return 0.0.0.0 because can't found any DNS setting.

Please add DNS

martin-ger, 2018-12-09:
Don't undertstand the problem: a DNS address is distributed by the AP. It redistributes the DNS it receives from the uplink AP. If none available (yet), it uses Google's 8.8.8.8.

wieb18, 2018-12-10:
Sorry... my mistake not explaining the details completely.

My network address is 192.168.0.0 and my router IP address is 192.168.0.230. I have ESP-12F as repeater (installed with your esp_wifi_repeater) which has IP address 192.168.0.1. Also I have ESP32 which connected to ESP-12F repeater.

**1.  DHCP**
With IP address from DHCP, ESP32 doesn't have an issue with DNS. DNS works perfectly. 

ESP32 got IP address 192.168.0.2 from ESP-12F DHCP. Although IP address 192.168.0.2 in range of my network address, but ESP32 can not be found. It seems ESP-12F repeater created its own network, that is why ESP32 can not detected in my network. My network only found ESP-12F repeater (192.168.0.1) and can not found ESP32.

**2. Fixed IP address**
Try ESP32 with fixed IP address 192.168.0.240. Same as #1 above, also can not found in my network (because ESP-12F repeater created separate netwok). And DNS not working in this mode, www.google .com got IP address 0.0.0.0. 

Those 2 issues not happen when I use real hardware repeater (TP-Link WA860RE).
1. Hardware repeater and ESP32 both found in my network range
2. No problem with DNS. With IP address from DHCP or with fixed IP address, both always has DNS setting from hardware repeater.

Please let me know, how to fix those 2 issues? 
I need both ESP-12F and ESP32 found in my network and receive DNS setting.

martin-ger, 2018-12-10:
The esp_wifi_repeater is in fact a NAT router as described in the Readme. This means it has two different IP network segments and does routing in between them. A node behind the AP cannot be reached directly from the STA side, even if it has an address of the range of the STA side (which is a strange/wrong IP config).

Do the following config:
- repeater has an IP address at the STA side, let's say 192.168.0.100
- repeater has an IP address at the AP side, let's say 192.168.4.1 (default)
- the ESP32 connected to the AP has a fixed IP, let's say 192.168.4.240

Now configure a portmap on the repeater:
portmap add TCP 80 192.168.4.240 80
Now you can reach the ESP32 (on port 80) from the outside (192.168.0.0/24 network) at the address of the repeater, i.e. 192.168.0.100. Configure and save this port mapping for any port you need to be accessible.

Hope this explains.

James0329, 2019-06-05:
I got same problem like this issue but have little different.
Here is my situation:
Hardware router at 192.168.1.1 with ssid "WIFI" and config local dns server 192.168.1.254
Local http server connect to hardware router and ip 192.168.1.200
Local dns server 192.168.1.254 running dnsmasq,have a record like "www.example.com=192.168.1.200"
Esp repeater connect to "WIFI" and get STA dhcp ip like 192.168.1.100 from hardware router.
ComputerA connect to esp repeater get dhcp ip like 192.168.4.2
If computerA request a website address "www.example.com" it will point to 192.168.1.200?

martin-ger, 2019-06-05:
Yes, it should: The esp_wifi_repeater by default distributes the same DNS server to its STA as the one it learned from its uplink AP. Usually standard home routers implement their own DNS server, so probabyl 192.168.4.2 with use 192.168.1.245 as DNS and resolve "www.example.com" to192.168.1.200.

James0329, 2019-06-05:
So if here have 5 esp repeater and use automesh mode link like a chain.
It still work well?I know them will have a group like 10.24.*.*
Any other computer if connect to any one of chain node still learn dns server from uplink hardware router?
Still will point to local webserver when them request "example.com"?

martin-ger, 2019-06-05:
Yes, they all will use the 192.168.1.245 as DNS. As long a requests from somewhere in the chain go outbound to 192.168.1.xx or further to the Internet - no problem.

gsirettito, 2020-11-28:
Hi, nice work.
I have been using my esp8266 as a repeater for a while now and it works quite well, but now I am connecting my Esp to a wifi with proxy and that resolves the connections through 2 DNS, but the fact is that I can only configure one and I think that that's what gives me connection instability.
How can i fix this?

eKetan, 2021-12-27:
> The esp_wifi_repeater is in fact a NAT router as described in the Readme. This means it has two different IP network segments and does routing in between them. A node behind the AP cannot be reached [...]

In this example Repeater STN IP is 192.168.0.100 STA IP and Repeater AP IP is 192.168.4.1 and  portmap add TCP 80 192.168.4.240 80 . 
So any TCP request to  STN 192.168.0.100  will be routed  to 192.168.4.240 by 192.168.4.1 Repeater. 
And above example working fine as this has to connect to one TCP port .

But in case if have more than one STNs  on 192.168.4.1  then how it will work  ?
For example if have two STNS (ESP) connected on 192.168.4.1  one is 192.168.4.240 as above And other is 192.168.4.200 

So Please if you have some idea how this can be achieved  ? 
Thanks!!