martin-ger/esp_wifi_repeater #251: Help with ping

ljrob61, 2018-12-10:
[image]

The AP-ESP1-STA is in Automesh mode.
I can control my Home Automation with my iPhone connected to AP-ESP1 or AP-AirPort.
My Home Router is set to give the iPhone 10.0.0.100 address.
The ESP32 can ping 10.0.0.100 to find my iPhone and know when I'm home.

The ESP1 has given the iPhone 10.24.1.2 and I have saved the DHCP lease.
But the ESP32 is unable to ping 10.24.1.2

Please can you help me with the settings I can change on ESP1 so I can
ping iPhone 10.24.1.2

Thank you.

Im using an ESP8266 USB dongle. It has an sm LED connected to GPIO 4
On your next update, Is it possible that when you disable HW Factory Reset
with "set hw_reset 255" that no voltage is on GPIO 4

[image]

ljrob61, 2018-12-15:
Thank you for your reply.
Im now using Version V2.2.2

I can confirm that setting portmap add TCP 80 10.24.1.2 80
that I can reach my iPhone, With the iPhone running a small web server.
But that's not really what I want.

My ESP32 is running my own Home Automation Arduino sketch with a ping library.
I can ping my iPhone and get a reply when it has a 10.0.0.100 Address. When it
is dark the lights will turn on but I have to be inside my house. I want to extend
my WiFi by using your WiFi Repeater code and turn the lights on when Im outside.
Im unable get a ping reply from my iPhone when it is connected to the ESP8266 with
an Address of 10.24.1.2

I have been searching the internet to find as much information as I can so I
would not ask stupid questions. I understand that ping does not use TCP UDP
or port numbers It uses ICMP.

In your README documentation you talk about ping in Static Routes.
I have read this many times trying to understand how it works and if would
would work in my case.

I would appreciate any more help you could give me.

I would like to thank you for changing the behaviour of the HW-reset.
I would also want to apologise for my mistake. The LED on my ESP8266 USB dongle
is connected to GPIO 14 and not GPIO 4 When the dongle is connected the GPIO14
LED is not very bright but it is still on.

Thank you.

martin-ger, 2018-12-15:
You cannot use a portmap for ICMP, that's right. If you really want to use ping (i.e. ICMP), static routing is probably an option to consider. However, this requires, that the ESP32 knows, that it can reach the 10.24.x.x network behind the 10.0.0.1 router. I.e. it needs a static route to it. Don't think you have a static route option on the ESP32...

As the iPhone behind the esp_router has the router as default gateway the route back should work without additional route entry.

ljrob61, 2018-12-15:
Thank you for your help.
I will look into static routing, Maybe put another ESP8266 repeater before the ESP32