martin-ger/esp_wifi_repeater #254: build error

fenglaijun, 2018-12-14:
i had installed esp-open-sdk success,but i still can't build success.the log is
     undefined reference to "wifi_station_clear_enterprise_identity"...
     i had tried to define WPA2_PEAP 0,but it still don't work,it's upset.
     thank you in advance!

martin-ger, 2018-12-14:
This function is new with NONOS SDK 2.1. Either install the SDK with an SDK 2.1 or later as base (see open-sdk's Makefile) or simply remove that call (you cannot use WPA2 enterprise anyway). An install of the sdk that works for me is: https://github.com/xxxajk/esp-open-sdk

fenglaijun, 2018-12-14:
> This function is new with NONOS SDK 2.1. Either install the SDK with an SDK 2.1 or later as base (see open-sdk's Makefile) or simply remove that call (you cannot use WPA2 enterprise anyway). An install of the sdk that works for me is: https://github.com/xxxajk/esp-open-sdk
thanks you for help!i will try it out.

closed 2018-12-14 by fenglaijun