martin-ger/esp_wifi_repeater #258: Beginner question

dickmann, 2019-01-02:
Hi Martin,

thanks for the great documentation you did.

I have a "Wireless module NodeMcu v3 v2 ESP32 D1MINI Lua WIFI development board ESP8266 with pcb Antenna and usb port ESP-12E CH34". On this I successfully installed your sketch as described, and this "repeater" works fine. Now I want to use another ESP32 (same model) to connect to the new repeater offering a webpage on port 80. 

Version V2.2.4 (build: Sat Dec 22 21:00:06 2018)
STA: SSID:SSIDHOMENET PW:***
BSSID: 08:96:d7:77:76:8f
Automesh: on (operational) Level: 1 Threshold: -85
AP:  SSID:SSIDHOMENET PW:*** IP:10.24.1.1/24 [NAT]
STA MAC: 60:01:94:2d:85:f3
AP MAC:  24:24:01:eb:30:fc
STA hostname: ESP_2d85f2
Network console access on port 7777 (mode 3)
Clock speed: 80
MQTT: disabled
CMD>

I want to connect from my browser to the port 80 of the connected ESP, any idea why this is not working?!?

I am also confused to read that the wifi repeater sw is working with 8266, but not with ESP32. My Lolin seems to be both, is this the reason? But just connecting via Laptop to the repeater works well.

Confused :-)
Any hint is highly appreciated!

Kind regards 
Michael

dickmann, 2019-01-03:
Thanks for your superfast response, now it works like a charm :-)

Any idea how I teach my system the hostname of the 10.24.1.2 device? I would like to be able to type in my browser "tempsensor" (the hostname) and he should connect to the device? I tried to google but I do not know what I am looking for, so any hint appreciated. So much to learn for me :-)

Big "Thank you"!

P.S.: a 1$ wifi repeater, how cool is that!

martin-ger, 2019-01-03:
Hi, you need to add a static entry in your /etc/hosts - on Windows:

https://blog.kowalczyk.info/article/10c/local-dns-modifications-on-windows-etchosts-equivalent.html