martin-ger/esp_wifi_repeater #259: Use Bridge mode with DHCP

happyTomatoe80, 2019-01-09:
Hi there!
Is it possible to use bridge mode with DHCP turned on?

martin-ger, 2019-01-10:
Don't understand exactly, what you are looking for. DHCP is working in both networks: AP and STA but as these are two networks, these are separate services with different addresses. If you are looking for a single network and address space- no this doesn't work with a router.

happyTomatoe80, 2019-01-13:
Would you be able to help me with instructions on how to configure bridge mode? I tried using Automesh after assigning a Static IP to the STA interface but, that did not seem to work. If I connect a device like a smartphone I can ping the esp and I can ping the main wifi device that the STA is extending, but I can't ping any other devices in the network. The main wifi device does not have dhcp and dns.

martin-ger, 2019-01-13:
Without individual portmaps you cannot reach a station behind the AP interface from the STA side - this is a limitation of NATing.