martin-ger/esp_wifi_repeater #261: make issue

m50008, 2019-01-13:
Hey, i get this error when trying to make the esp_wifi_repeater.
/home/user/esp-open-sdk/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: build/app.out section `.text' will not fit in region `iram1_0_seg'
collect2: error: ld returned 1 exit status
Makefile:132: recipe for target 'firmware/0x02000.bin' failed
make: *** [firmware/0x02000.bin] Error 1
Can someone help me?

martin-ger, 2019-01-13:
Did you try the esp-open-sdk with SDK 2.2 e.g. from https://github.com/xxxajk/esp-open-sdk. If not, try to disable 

#define	WPA2_PEAP 1

in L 137 of user_config.h.

m50008, 2019-01-14:
I have the latest esp-open-sdk.
When disabling #define WPA2_PEAP 1 the error stays.

gr01d, 2019-01-21:
try to disable some features from user_config.h.

ken9674, 2020-06-03:
I am having trouble with the same error when "#define HAVE_ENC28J60" is set to 1.
Please help me.

gr01d, 2020-06-03:
Do you need the ENC28J60 functionality? If not then comment it out to disable it. Also, check the recommended esp_open_sdk package for this project, the latest ones may/may not work.

ken9674, 2020-06-04:
I could compile without ENC28J60.
But I want to configure a router with ENC28J60.
I recently installed my sdk.
My version may be the cause of the error.
Can I compile with a simple fix?

gr01d, 2020-06-13:
Try different version of the sdk, have you modified the project files? if yes, then try to compile the upstream 'master' branch if it works.