martin-ger/esp_wifi_repeater #267: How set STA static IP & gateway ip

easygetesp, 2019-01-25:
I try to change sta ip (generated by dhcp) by TCP 777 client command set ip @ set gw but not working. Please help for how change ip.

closed 2019-01-27 by martin-ger

putin007, 2019-08-31:
I plan to make a mesh network on + 20pcs esp8266
how to set stats in mesh mode?
for all network levels - for each esp8266 its own ip address is about 192.168.1.x
 
please give a complete sequence of commands.
where can i read this?

thank you so much for this project
dream to tie repeater and home assistant (esphome)
all the best to you!

martin-ger, 2019-08-31:
Try the automesh mode. Only ssid and pw required.

12368w, 2019-08-31:
> […](#)

I want to make simple relay control on every esp8266 in a mesh network
I realized that I can do a redirect from the internal network, for example 10.24.1.x -> 192.168.1.x - but I did not find a good description of the sequence of commands

thank!