martin-ger/esp_wifi_repeater #270: how to recompile source code on Windows?

kitty801, 2019-02-04:
how to recompile source code on Windows?

martin-ger, 2019-02-04:
It might  work with the PlatformIO environment, but this is not supported by the current repo.

robnewton, 2019-02-19:
I had success on Windows by simply using a VirtualBox image of Ubuntu. Compile of the toolchain took about an hour though so you might give yours a bit more than 1Gig ram like I did.

A good source is https://www.osboxes.org/ubuntu/

I used the 18.10 image and it worked fine.

NightFalcon650, 2022-06-05:
> I had success on Windows by simply using a VirtualBox image of Ubuntu. Compile of the toolchain took about an hour though so you might give yours a bit more than 1Gig ram like I did.
> 
> A good source is https://www.osboxes.org/ubuntu/
> 
> I used the 18.10 image and it worked fine.

Could you wright a simple tutorial what you did on the Ubuntu box. What did you install, what commands?
Or could you put that in a tekst file or script?