martin-ger/esp_wifi_repeater #272: How to change acces point name?

kitty801, 2019-02-07:
How to change acces point name?
I've tried to change WIFI_AP_SSID "MyAP" entry in user\user_config.h and then recompile project, but acces point is always MyAP.

martin-ger, 2019-02-07:
Try a "reset factory" on an ESP that had the original version installed before.

kitty801, 2019-02-09:
It doesn't helps.
After factory reset and flash my recompiled bin , acces point is MyAP 
even though I entered a different name in the **user\user_config.h** file.
Do I need to recompile some other files in esp-open-sdk dir?

kitty801, 2019-02-09:
Also I'v tried to change some texts in user\web.h file and I don't see my changes too.

robnewton, 2019-02-15:
I ran into that issue and had to erase then reflash before it was fixed. Didn't look into root cause.

kitty801, 2019-02-18:
I've found what was problem.
Before every make command, I have to delete **build** dir.

martin-ger, 2019-02-19:
The ESP saves its last WiFi config in flash in the high memory blocks. It restarts with the latest config. If you either write a new one or erase the complete memory it will be gone.

With the WiFi repeater you can either do a "reset factory" via CLI (same as pulling the pin down) or simply configure a new one and do "save" "reset"

kitty801, 2019-02-21:
But I don't want to unplug my repeater and plug to computer to send factory reset command.
I'd like to do it wireless.

martin-ger, 2019-02-22:
You can log into the ESP: with "telnet 7777 192.168.4.1",  on windows use putty. Now you can enter all commands incl. "reset factory".

kitty801, 2019-02-23:
I'm connected to my network (but my ESP works in mesh mode , so I don't know if connection is via ESP or directly to router).
Then I try to log into the ESP with "telnet 7777 192.168.4.1" (use putty) but I can't connect.
I have to connect via ESP?

gordoste, 2019-06-13:
He made a typo. Use "telnet 192.168.4.1 7777"

closed 2019-07-22 by martin-ger