martin-ger/esp_wifi_repeater #273: Externalize user config

robnewton, 2019-02-08:
Great project! I will flash 200 esp's and need a predefined serial number included in each AP name and I'm thinking about how to automate this. 

I have considered... 
- automating compilation with changes to user config between each compile
- powering the esp after flashing it then use telnet
- externalize the settings to a file which can be burned to the esp but not compiled. It would need to be read by the firmware at boot

Do you have any thoughts on the best approach?

martin-ger, 2019-02-08:
A unique AP name can be derived from the individual MAC address. This would require minor changes in the code.

Another option, especially if you not just want to have unique, but freely defined names, would be to write a script to flash the config. Config data is written to flash block 0x60, i.e. address 0x60000. The structure of this data is given in config_flash.h. "ap_ssid" is the value you want to change. This means: configure one ESP and read the 0x2000 byte starting from 0x60000 with the esptool. Than after firmware flash patch the new ap_ssid into this binary file and write it to 0x60000.

robnewton, 2019-02-08:
Oh yes I like the patch idea. I think I'll take that approach. Thanks!

closed 2019-07-22 by martin-ger