martin-ger/esp_wifi_repeater #275: Building on mac osx

robnewton, 2019-02-15:
I'm trying to follow the readme at https://github.com/xxxajk/esp-open-sdk for setting up the build environment on OSX however I am running into errors. Has anyone set this up to compile on OSX recently that could provide a novice some more up-to-date instructions? I do not use OSX or Linux/Unix very often but would like to test the recent GPIO code I've added.

For instance...
`$ brew tap homebrew/dupes
Error: homebrew/dupes was deprecated. This tap is now empty as all its formulae were migrated.`

and...
`$ brew install binutils coreutils automake wget gawk libtool help2man gperf gnu-sed --with-default-names grep
Error: invalid option: --with-default-names
`

robnewton, 2019-02-16:
As a start to try and get around these issues, I have upgraded homebrew, then removed the invalid option from the brew install command based on the following: (it was removed)

https://github.com/Homebrew/homebrew-core/commit/8c4f31e6643469493fc26d275be09cb05b09beac

robnewton, 2019-02-16:
Created and mounted the case sensitive volume. Cloned the SDK repo. CD to the sdk dir then typed make. Got the following error and fixed it with the following as well based on this: https://github.com/pfalcon/esp-open-sdk/issues/350

The error:
`missing required tool: gobjcopy objcopy`

Execute this to fix it:
`export PATH="/usr/local/opt/binutils/bin:$PATH"`

robnewton, 2019-02-16:
Make made it pretty far but ultimately fails installing ncurses for build...

```
...
***********************************************************

Now configured for "xtensa-lx106-elf"
sed -r -i.org s%CT_PREFIX_DIR=.*%CT_PREFIX_DIR="/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf"% .config
sed -r -i s%CT_INSTALL_DIR_RO=y%"#"CT_INSTALL_DIR_RO=y% .config
cat ../crosstool-config-overrides >> .config
./ct-ng build
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20190216.151147
[INFO ]  Building environment variables
[INFO ]  =================================================================
[INFO ]  Retrieving needed toolchain components' tarballs
[INFO ]  Retrieving needed toolchain components' tarballs: done in 59.00s (at 01:03)
[INFO ]  =================================================================
[INFO ]  Extracting and patching toolchain components
[INFO ]  Extracting and patching toolchain components: done in 149.00s (at 03:32)
[INFO ]  =================================================================
[INFO ]  Installing ncurses for build
[ERROR]    clang: error: linker command failed with exit code 1 (use -v to see invocation)
[ERROR]    make[4]: *** [clear] Error 1
[ERROR]    clang: error: linker command failed with exit code 1 (use -v to see invocation)
[ERROR]    make[4]: *** [tabs] Error 1
[ERROR]    make[3]: *** [all] Error 2
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing ncurses for build'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@216]
[ERROR]  >>        called from: do_ncurses_backend[scripts/build/companion_libs/220-ncurses.sh@151]
[ERROR]  >>        called from: do_ncurses_for_build[scripts/build/companion_libs/220-ncurses.sh@35]
[ERROR]  >>        called from: do_companion_libs_for_build[scripts/build/companion_libs.sh@29]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6f/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 4:22.00)
[04:23] / make[2]: *** [build] Error 1
make[1]: *** [_toolchain] Error 2
make: *** [crosstool-NG/.built] Error 2
```

robnewton, 2019-02-16:
Unfortunately it looks as though Crosstool-ng no longer supports OSX:

https://crosstool-ng.github.io/2018/11/26/macos.html

robnewton, 2019-02-18:
Never could get OSX working however I was able to get up and running quickly with an Ubuntu VirtualBox  image from osboxes.org

closed 2019-02-18 by robnewton

orn-a, 2020-09-29:
```
brew install ncurses
export CPPFLAGS="-I/usr/local/opt/ncurses/include -I/usr/local/opt/gettext/include"
export LDFLAGS="-L/usr/local/opt/gettext/lib -lintl -L/usr/local/opt/ncurses/lib"
```
fixes this for me