martin-ger/esp_wifi_repeater #279: ESP don't connevt to uplink AP

Bricator, 2019-03-03:
I'm trying to use an ESP8266 as wifi repeater.
My computer links OK to the AP side of the ESP at address x.x.4.1
I put ssid and password.
Then : 
If I connect again to AP, the form is empty and I must enter again ssid and password.
The main problem is ESP fails to connect to the uplink AP.
From the connected station by x.x.4.1 I can telnet to ESP, redefining ssid and password.
With "show stats" I can see the phy_mode is "n". My acces point is b and g, and this can be the reason.
I put "set phy_mode 2", but then the connection is lost and the next time nothing is changed.

I tried by serial console, and I get repeatedly in the console the lines : 
  -reconnect
  -state: 2 -> 0 (0)
  -scandone
  -state: 0 -> 2 (b0)
  -state: 2 -> 3 (0)
  -state: 3 -> 5 (10)
  -add 0
  -aid 1
  -cnt
  -state: 5 -> 2 (fc0)
  -rm 0
  -disconnect from ssid xxx, reason 15
I "set ssid xxx" and "set password xxx" (and now the form is filled when I access by the web page)
And if I try "show ..." the call appears in the list, but the answer is lost
I tried in the console "set phy_mode 2", but can't see any change nor any answer

What happens on the console when I click "connect" in the web page : 
............................
disconnect from ssid xxx, reason 15
--here I click "connect" on the web page--
Saving configuration
Restarting ... 
state: 2 -> 0 (0)
vPortFree 515 line assert fail
del if0
station: (Uplink AP MAC) leave, AID = 1
rm 1
vPortFree 515 line assert fail
bcn 0
del if1
usl

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x40100000, len 1328, room 16 
tail 0
chksum 0x12
load 0x3ffe8000, len 604, room 8 
tail 4
chksum 0x34
csum 0x34

rBoot v1.4.2 - richardaburton@gmail.com
Flash Size:   32 Mbit
Flash Mode:   DIO
Flash Speed:  80 MHz
Booting rom 0.

--garbage--

Repeater V2.2.6 starting
running rom 0
Config found and loaded
Starting Console TCP Server on port 7777
Starting Web Config Server on port 80
mode : sta(3c:71:bf:29:63:4a) + softAP(3e:71:bf:29:63:4a)
add if0
add if1
dhcp server start:(ip:xxx.xxx.4.1,mask:255.255.255.0,gw:xxx.xxx.4.1)
bcn 100
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
add 1
aid 1
station: (Uplink AP MAC) join, AID = 1
station: (Uplink AP MAC) join, AID = 1
state: 5 -> 2 (fc0)
rm 0
disconnect from ssid xxx, reason 15
reconnect
state: 2 -> 0 (0)
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
state: 5 -> 2 (fc0)
rm 0
disconnect from ssid xxx, reason 15
reconnect
............................

"reason 15" : what can I do ?
I have MAC accesse filter but the ESP STA MAC is allowed (or I get other errors 2, 3, 201, 203, 204)

Thnk you very much ... and mainly for this promissing (for me) work

martin-ger, 2019-03-03:
Reason 15 is: 
`REASON_4WAY_HANDSHAKE_TIMEOUT   = 15,  /* 11i */`
Something is wrong/not matching in the WiFi communication. Don't think it is a problem of the repeater SW. Does your ESP connect to the WiFi when running other programs (e.g. Arduino)?

The special thing here seems to be the MAC filter. Do you really allow 3c:71:bf:29:63:4a? Is it possible to disable the filter for a test?

Bricator, 2019-03-03:
Thank you for answer.
When disabling the MAC filter I get on beginning "reason 4" or "reason 15" and after "reason 15".
I tried enabling 3e:71... (soft AP MAC insted of STA MAC) and then received these errors, 2, 3, 201, 203 ...
Yes I think the issue comes from my configuration. What about phy_mode ? I didn't succeed in changing it (my AP is only b and g). When I try from telnet on x.x.4.1 7777, the link is broken so no possibility to "save".
With another ESP there is no problem connecting with this AP.
The soft AP side works well.
Thank you

Bricator, 2019-03-03:
I tried on another AP, wich is open without any protection and I got in the console : 
  -reconnect
  -disconnect from ssid xxx, reason 2
  -scandone
  -state 0 -> 2 (b0)
  -state 2 -> 0 (2)
  -reconnect
  -...

martin-ger, 2019-03-04:
Strange - guess, this should not happen on an open AP. These are the reason codes:
```
REASON_UNSPECIFIED              = 1,
   REASON_AUTH_EXPIRE              = 2,
   REASON_AUTH_LEAVE               = 3,
   REASON_ASSOC_EXPIRE             = 4,
   REASON_ASSOC_TOOMANY            = 5,
   REASON_NOT_AUTHED               = 6,
   REASON_NOT_ASSOCED              = 7,
   REASON_ASSOC_LEAVE              = 8,
   REASON_ASSOC_NOT_AUTHED         = 9,
   REASON_DISASSOC_PWRCAP_BAD      = 10,  /* 11h */
   REASON_DISASSOC_SUPCHAN_BAD     = 11,  /* 11h */
   REASON_IE_INVALID               = 13,  /* 11i */
   REASON_MIC_FAILURE              = 14,  /* 11i */
   REASON_4WAY_HANDSHAKE_TIMEOUT   = 15,  /* 11i */
   REASON_GROUP_KEY_UPDATE_TIMEOUT = 16,  /* 11i */
   REASON_IE_IN_4WAY_DIFFERS       = 17,  /* 11i */
   REASON_GROUP_CIPHER_INVALID     = 18,  /* 11i */
   REASON_PAIRWISE_CIPHER_INVALID  = 19,  /* 11i */
   REASON_AKMP_INVALID             = 20,  /* 11i */
   REASON_UNSUPP_RSN_IE_VERSION    = 21,  /* 11i */
   REASON_INVALID_RSN_IE_CAP       = 22,  /* 11i */
   REASON_802_1X_AUTH_FAILED       = 23,  /* 11i */
   REASON_CIPHER_SUITE_REJECTED    = 24,  /* 11i */

   REASON_BEACON_TIMEOUT         = 200,
   REASON_NO_AP_FOUND              = 201,
   REASON_AUTH_FAIL            = 202,
   REASON_ASSOC_FAIL            = 203,
   REASON_HANDSHAKE_TIMEOUT      = 204,
```

Bricator, 2019-03-04:
Thank you so much for this list that will be useful.
But the mystery will stay : this morning, all is well.
After connecting to my open AP as soon it was on, I tried the secured one and it works. I'm even now accessing to internet through the repeater and the secured AP !

In fact I had few actions : 
Initialy, I saved the original firmware and then wiped off the full memory before loading esp_wifi_repeater.
So I restored the initial firmware, and without erasing the memory reloaded esp_wifi_repeater. The reason ? 

Thank you so much again for your answers, I'm so sorry ...
And thank you for this wifi repeater.

Bricator, 2019-03-14:
I should have liked an answer to my last question.
Anyway, thank you for help.

closed 2019-03-14 by Bricator

martin-ger, 2019-03-14:
Actually, I don't know an answer, as I cannot reproduce this behavior.

kock-rueil, 2019-08-20:
I meet the same issue with the same console output using two different routers. ESP firmware seems to be responsible. How can i help to solde this issue ? Daniel.

raipat, 2020-04-04:
experiencing the same problem with a ESP-8286MOD 12F sold as d1 mini clone from az-delivery
I found your awesome project through their website.

from console:
WiFi Repeater V2.2.15 starting

running rom 0
Config found and loaded (3216 Bytes)
Starting Console TCP Server on port 7777
Starting Web Config Server on port 80
mode : sta(ec:fa:bc:6e:f3:c4) + softAP(24:24:01:08:21:e0)
add if0
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt
state: 5 -> 2 (2a0)
rm 0
disconnect from ssid HOME, reason 2
Initial connect to SSID HOME failed, check password - factory reset

password is correct, the AP is a Fritz!Box 7590. I tried both guest and normal net. No mac filters.
Also tried the hotspot of my iPhone, same problem, failed with reason 2.

raipat, 2020-04-04:
seems to be a defective chip. I have 4 of those d1 mini clones.
Just flashed the same firmware binary onto a second unit and it works instantly.

raipat, 2020-04-04:
3 out of 4 do work