martin-ger/esp_wifi_repeater #281: possibility of adding program in parallel

NHKhaled, 2019-03-11:
Hello,
In fact, I am a noob in Arduino / ESP projects, but I am very interested in your project.

My question is: Is it possible to add an internal program to this esp?
I mean, in addition to a repeater function, is it possible to add a program with Arduino IDE such as using GPIOS, MQTT_sub_pub, etc .. ?

martin-ger, 2019-03-13:
It is possible to extend the functionality using the esp-open-sdk. Perhaps you first have a look whether the program already supports what you need. There are quite a lot options for GPIOs and MQTT. Using the "command" MQTT topic and the "gpio" command you can e.g. set and read GPIO pins...

Using it in Arduino is somewhat hard. You might try to replace the liblwib there with the one from my repo: https://github.com/martin-ger/esp-open-lwip . Then you can use the NAT functionality plus your own code. However, this hasn't been done so far to my knowledge, and as you describe yourself as a noob, it is probably not the first thing you want to try...

martin-ger, 2019-03-16:
Look for this: https://github.com/martin-ger/lwip_nat_arduino

closed 2019-03-16 by martin-ger